awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #553: [QUESTION] How to Carry Per-frame Metadata Along With Media?

hector-chou, 2020-08-25:
I have per-frame (e.g. VCA) metadata to carry, but I don't know how to do it in KVS.

KinesisVideoFrame has no room for this, and addTrack() doesn't seem to support adding custom tracks.
The metadata changes frame by frame, so per-fragment metadata won't fit my case.

In fact, I don't know is it even achievable in KVS. Any idea?

MushMal, 2020-08-25:
Interesting case. True, it’s a format and technology limitation to have fragment level metadata only. We currently enforce total of 10 (need to double check) metadata per fragment on the backend.
Here are some ideas:

1) have a single metadata that carries information for all of the frames in a special format that can be mapped back to the frames. You could use frame time codes or the sequence of those to do the mapping.
2) have a custom frame format where you capture the metadata together with the frame data. Note, your consumer will have to understand the format then.
3) have a separate track that outputs special frame format for your metadata that you can map to the frames by using frame time code or simply the sequence number. Currently, there are 3 tracks limit but we will be extending it.
4) have a separate stream that streams out the metadata in a custom format your consumer will understand. The main issue here is that your consumers will have to do av/sync to map the timestamps of the metadata stream to the main stream

hector-chou, 2020-08-25:
Thank you for replying so fast!

> 1. have a single metadata that carries information for all of the frames in a special format that can be mapped back to the frames. You could use frame time codes or the sequence of those to do the mapping.
This is also the first approach comes up to my mind, but there are number and size limitations on fragment metadata:
- You can prepend up to 10 metadata items to a fragment.
- A fragment metadata name can be up to 128 bytes in length.
- A fragment metadata value can be up to 256 bytes in length.
I'm afraid that VCA metadata won't fit these limits.

> 2. have a custom frame format where you capture the metadata together with the frame data. Note, your consumer will have to understand the format then.
That would be my last resort.
By the way, will this approach affect KVS HLS output in any ways?

> 3. have a separate track that outputs special frame format for your metadata that you can map to the frames by using frame time code or simply the sequence number. Currently, there are 3 tracks limit but we will be extending it.
I do want to have 3 tracks: video, audio, and metadata. However, addTrack() demands "codec ID" and "track type" as parameters, but I don't see any reasonable value to pass in for the metadata track.
Or can I sneak it in by declaring it as a video track, and put metadata as frameData when I call putFrame()?

MushMal, 2020-08-25:
HLS/DASH won’t work with custom frame format as it has to transpackage the MKV to mpeg.

You can specify some custom name for the track and use putframe. Really need to double check if this will affect HLS/DASH at all, unfortunately they are fairly restrictive currently

hector-chou, 2020-08-25:
> HLS/DASH won’t work with custom frame format as it has to transpackage the MKV to mpeg.

Understood.

> You can specify some custom name for the track and use putframe.

`addTrack(const uint64_t track_id, const string &track_name, const string &codec_id, MKV_TRACK_INFO_TYPE track_type, ...)`

Custom "track_name" would be okay, but how about "codec_id" and "track_type"? Wouldn't them affect anything if I just pass a random string/integer in?

Especially "track_type", there is no enum defined for custom track:

`typedef enum { MKV_TRACK_INFO_TYPE_VIDEO = (BYTE) 0x01, MKV_TRACK_INFO_TYPE_AUDIO = (BYTE) 0x02, MKV_TRACK_INFO_TYPE_UNKOWN = (BYTE) 0x03,} MKV_TRACK_INFO_TYPE, *PMKV_TRACK_INFO_TYPE;`

MushMal, 2020-08-25:
You can specify your own codec_id and the track_type can be MKV_TRACK_INFO_TYPE_UNKOWN

hector-chou, 2020-08-26:
Thank you very much! I'm going to close this issue and give it a try.

closed 2020-08-26 by hector-chou

qraleq, 2023-11-29:
Hi @hector-chou, we're working on something very similar to your use-case described in this issue. I'm wondering if you ever managed this solution to work for you.