awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #554: How to stream multiple cameras (producers)-RTSP in AWS Kinesis Video Stream using gstreamer

bhavikapanara, 2020-08-26:
Hi, There

I have successfully got a single RTSP camera stream to AWS kinesis video stream using GStreamer. Below is the command I am using for this.

`gst-launch-1.0 rtspsrc location="rtsp://.../live01_rec" short-header=TRUE ! rtph264depay ! video/x-h264, width=640,height=480,framerate=15/1, format=avc,alignment=au ! kvssink stream-name="MyTestStream" storage-size=512 access-key="AKIA..." secret-key="Q+oD..." aws-region="us-east-2"`

But, How we can stream multiple cameras simultaneously?

For multiple camera streaming, do I need to create multiple AWS Kinesis video streams or I can achieve it using single kinesis video stream?

Somewhere, I found the below command to stream multiple cameras but it doesn't work. 

`gst-launch-1.0 rtspsrc location="rtsp://.../live01_rec" location="rtsp://.../live02_rec" short-header=TRUE ! rtph264depay ! video/x-h264, width=640,height=480,framerate=15/1, format=avc,alignment=au ! kvssink stream-name="MyStream_01" kvssink stream-name="MyStream_02" storage-size=512 access-key="AKIA..." secret-key="Q+oD..." aws-region="us-east-2"`

Can someone please help me to solve my issue.

Thanks,
Bhavika

MushMal, 2020-08-26:
The kvs plugin itself is not designed to operate on multiple streams so you must either use separate gst-launch commands with each stream or have some form of a parallelization but I am not sure how exactly that's done in the GStreamer syntax. Perhaps gstreamer forums are the right choice. NOTE: you can try to use fakesink to get your pipeline working and then replace it with kvssink to make it easier to debug.

closed 2020-09-11 by bhavikapanara

simonovmaks, 2021-11-19:
"rtsp://.../live01" have you tried this link?

Madesh1997, 2022-01-28:
@bhavikapanara Hi, did you find anyway to stream video from multiple camera?