awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #555: raspberry pi No such element or plugin 'kvssink' help me!!

MAKER-park, 2020-08-26:
A one liner description about the use case and what you are trying to achieve

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

hello! i try your git read me text install gst, cmake, byacc flex etc...

and git clone your git repo and make build 

and,,, cmake & make all command 
it is result

pi@raspberrypi:~/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build $ sudo cmake ..
-- Kinesis Video Cpp Producer path is /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No configure step for 'libkvscproducer-download'
[ 33%] No build step for 'libkvscproducer-download'
[ 44%] No install step for 'libkvscproducer-download'
[ 55%] No test step for 'libkvscproducer-download'
[ 66%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- openssl already built
-- curl already built
-- Finished building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No configure step for 'libkvspic-download'
[ 33%] No build step for 'libkvspic-download'
[ 44%] No install step for 'libkvspic-download'
[ 55%] No test step for 'libkvspic-download'
[ 66%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
-- Found OpenSSL: /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so (found version "1.1.1g")
-- Bitness 32 bits
-- Found OpenSSL: /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so (found suitable version "1.1.1g", minimum required is "1")
found log4cplus.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build
pi@raspberrypi:~/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build $ sudo make all
[  1%] Built target kvs_gstreamer_file_uploader_sample
[ 37%] Built target kvspic
[ 57%] Built target kvspicUtils
[ 66%] Built target kvsCommonCurl
[ 77%] Built target cproducer
[ 86%] Built target KinesisVideoProducer
[ 91%] Built target gstkvssink
[ 93%] Built target kvs_gstreamer_sample
[ 94%] Built target kvs_gstreamer_multistream_sample
[ 97%] Built target KinesisVideoProducerJNI
[100%] Built target kvs_gstreamer_audio_video_sample
pi@raspberrypi:~/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build $ sudo gst-inspect-1.0 kvssink
No such element or plugin 'kvssink'

how to make sure install this aws kinesis! it's very difficult to me.
so Can you give me some advice?

MushMal, 2020-08-26:
Make sure you install the GStreamer prerequisites per readme: on nixes it's 
`$ sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
`

Next, per readme https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#loading-element

you need 

```
export GST_PLUGIN_PATH=`pwd`/build
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
```

disa6302, 2020-08-26:
One more thing. Do you run cmake .. -DBUILD_GSTREAMER_PLUGIN=ON? This needs to be run if you want the kvssink library to be generated.

MAKER-park, 2020-08-27:
thank you!! i solve it is problem export GST_PLUGIN_PATH=`pwd`/build
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib !
but

MushMal, 2020-08-27:
Your access key/secret key combination is invalid. Make sure you specify the correct one

MAKER-park, 2020-08-28:
thank you!! i retry it's done! 
but.. i faced other problem... 
try python script 
    subprocess.call("gst-inspect-1.0 kvssink", shell=True)
like this but No such element or plugin 'kvssink' ....
but command is all right... why this happened... it is very confused me...

MAKER-park, 2020-08-28:
i check python script gst-inspect-1.0
pi@raspberrypi:~/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp $ sudo python command_test.py 
video4linux2:  v4l2src: Video (video4linux2) Source
video4linux2:  v4l2sink: Video (video4linux2) Sink
video4linux2:  v4l2radio: Radio (video4linux2) Tuner
video4linux2:  v4l2deviceprovider (GstDeviceProviderFactory)
video4linux2:  v4l2jpegdec: V4L2 JPEG Decoder
video4linux2:  v4l2mpeg4dec: V4L2 MPEG4 Decoder
video4linux2:  v4l2h263dec: V4L2 H263 Decoder
video4linux2:  v4l2h264dec: V4L2 H264 Decoder
video4linux2:  v4l2vp8dec: V4L2 VP8 Decoder
video4linux2:  v4l2h264enc: V4L2 H.264 Encoder
video4linux2:  v4l2convert: V4L2 Video Converter
gtk:  gtksink: Gtk Video Sink
gtk:  gtkglsink: Gtk GL Video Sink
opengl:  glimagesink: GL Sink Bin
opengl:  glimagesinkelement: OpenGL video sink
opengl:  glupload: OpenGL uploader
opengl:  gldownload: OpenGL downloader
opengl:  glcolorconvert: OpenGL color converter
opengl:  glcolorbalance: Video balance
opengl:  glfilterbin: GL Filter Bin
opengl:  glsinkbin: GL Sink Bin
opengl:  glsrcbin: GL Src Bin
opengl:  glfiltercube: OpenGL cube filter
opengl:  gltransformation: OpenGL transformation filter
opengl:  glvideoflip: OpenGL video flip filter
opengl:  gleffects: Gstreamer OpenGL Effects
opengl:  gleffects_identity: Do nothing Effect
opengl:  gleffects_mirror: Mirror Effect
opengl:  gleffects_squeeze: Squeeze Effect
opengl:  gleffects_stretch: Stretch Effect
opengl:  gleffects_tunnel: Light Tunnel Effect
opengl:  gleffects_fisheye: FishEye Effect
opengl:  gleffects_twirl: Twirl Effect
opengl:  gleffects_bulge: Bulge Effect
opengl:  gleffects_square: Square Effect
opengl:  gleffects_heat: Heat Signature Effect
opengl:  gleffects_sepia: Sepia Toning Effect
opengl:  gleffects_xpro: Cross Processing Effect
opengl:  gleffects_lumaxpro: Luma Cross Processing Effect
opengl:  gleffects_xray: Glowing negative effect
opengl:  gleffects_sin: All Grey but Red Effect
opengl:  gleffects_glow: Glow Lighting Effect
opengl:  gleffects_sobel: Sobel edge detection Effect
opengl:  gleffects_blur: Blur with 9x9 separable convolution Effect
opengl:  gleffects_laplacian: Laplacian Convolution Demo Effect
opengl:  glcolorscale: OpenGL color scale
opengl:  glshader: OpenGL fragment shader filter
opengl:  glfilterapp: OpenGL application filter
opengl:  glviewconvert: OpenGL Multiview/3D conversion filter
opengl:  glstereosplit: GLStereoSplit
opengl:  gltestsrc: Video test source
opengl:  gldeinterlace: OpenGL deinterlacing filter
opengl:  gloverlay: Gstreamer OpenGL Overlay
opengl:  glfilterglass: OpenGL glass filter
opengl:  gldifferencematte: Gstreamer OpenGL DifferenceMatte
pulseaudio:  pulsesink: PulseAudio Audio Sink
pulseaudio:  pulsesrc: PulseAudio Audio Source
pulseaudio:  pulsedeviceprovider (GstDeviceProviderFactory)
mpeg2enc:  mpeg2enc: mpeg2enc video encoder
cutter:  cutter: Audio cutter
isomp4:  qtdemux: QuickTime demuxer
isomp4:  rtpxqtdepay: RTP packet depayloader
isomp4:  qtmux: QuickTime Muxer
isomp4:  mp4mux: MP4 Muxer
isomp4:  ismlmux: ISML Muxer
isomp4:  3gppmux: 3GPP Muxer
isomp4:  mj2mux: MJ2 Muxer
isomp4:  qtmoovrecover: QT Moov Recover
[... 2785 lines cut ...]
videobox:  videobox: Video box filter
gme:  gmedec: Gaming console music file decoder
videoparsersbad:  h263parse: H.263 parser
videoparsersbad:  h264parse: H.264 parser
videoparsersbad:  diracparse: Dirac parser
videoparsersbad:  mpegvideoparse: MPEG video elementary stream parser
videoparsersbad:  mpeg4videoparse: MPEG 4 video elementary stream parser
videoparsersbad:  pngparse: PNG parser
videoparsersbad:  jpeg2000parse: JPEG 2000 parser
videoparsersbad:  h265parse: H.265 parser
videoparsersbad:  vc1parse: VC1 parser
chromaprint:  chromaprint: Chromaprint fingerprinting element
audioconvert:  audioconvert: Audio converter
vpx:  vp8dec: On2 VP8 Decoder
vpx:  vp8enc: On2 VP8 Encoder
vpx:  vp9dec: On2 VP9 Decoder
vpx:  vp9enc: On2 VP9 Encoder
cacasink:  cacasink: A colored ASCII art video sink
dc1394:  dc1394src: 1394 IIDC Video Source
staticelements:  bin: Generic bin
staticelements:  pipeline: Pipeline object

Total count: 250 plugins (4 blacklist entries not shown), 1436 features

it different 246 250 plugins... how can i do??

disa6302, 2020-09-17:
@MAKER-park ,

Not sure what your question is. Please choose the appropriate media source. I am closing this since the original issue is resolved
Feel free to reach out if you have any more questions

closed 2020-09-17 by disa6302