awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #556: blockingCurlCall(): Curl perform failed for second device registration

thasin96, 2020-08-28:
I had created an IOT device called "X" for kvs application. I got certificates for the device, using the certs, I was able to successfully start streaming. Now, I switched to a new AWS Account, and created a device called "Y". I followed the same steps to download certs for the device "Y". But, I am getting `blockingCurlCall(): Curl perform failed`  error.  It seems multiple device registration in single device in ca_cert_file causes this issue. 

How to fix the issue? 
I want to use my new certs to start streaming. 

Complete StackTrace
```

2020-08-28 07:34:30 ERROR   blockingCurlCall(): Curl perform failed for url https://credential-enppoint.credentials.iot.ap-southeast-1.amazonaws.com/role-aliases/test_role_alias/credentials with result Problem with the SSL CA cert (path? access rights?) : error setting certificate verify locations:
  CAfile: /home/thasin/certs/AmazonRootCA1.pem
  CApath: /etc/ssl/certs 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create Iot Credential provider. Error status: 0x15000011
Aborted
```

MushMal, 2020-08-28:
You should use the same root ca cert listed here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/certs but use the new certificate and it's endpoint.

Likely, you are intermixing the certs or endpoints. Please run a few scenarios.

disa6302, 2020-09-17:
Closing this since the issue has gone stale and looks like the question is answered. Feel free to reach out if you have any questions.

closed 2020-09-17 by disa6302