awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #557: [QUESTION]

tingz2004, 2020-08-29:
A one liner description about the use case and what you are trying to achieve
Running gst-launch-1.0  with built-in camera on MacBook Pro failed

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 
Running gst-launch-1.0 with mkv file succsee, but with built-in camera failed.
The output has following error:
ERROR: from element /GstPipeline:pipeline0/GstAutoVideoSrc:autovideosrc0/GstAVFVideoSrc:autovideosrc0-actual-src-avfvide: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): void gst_base_src_loop(GstPad *) (): /GstPipeline:pipeline0/GstAutoVideoSrc:autovideosrc0/GstAVFVideoSrc:autovideosrc0-actual-src-avfvide:
streaming stopped, reason not-negotiated (-4)

The gst-launch-1.0 command I used is
gst-launch-1.0 -v autovideosrc ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=500 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

MushMal, 2020-08-29:
Seems that your encoder/camera resolution or frame rate mismatch. Try playing with resolutions. Remove kvssink and use fakesink to validate the pipeline. Ask GStreamer forums for more help.

closed 2020-08-31 by tingz2004

KoStard, 2021-01-24:
Did you resolve the issue? Having the same problem.

MushMal, 2021-01-25:
@KoStard depending on your macbook pro, you might have resolutions that are not supported by your hardware encoder (vtenc). You could try to remove the bitrate param from the encoder and modify the resolution and frame rate for the video convert params.

More info: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/macos.md

KoStard, 2021-01-25:
Thank you for your reply @MushMal , I found a solution for my problem, sharing the working command, as it may be useful for others too 😄 

```
gst-launch-1.0 -v autovideosrc device=/dev/video0 \
! video/x-raw,framerate=20/1 \
! videoscale \
! videoconvert \
! x264enc tune=zerolatency bitrate=200 speed-preset=superfast \
! kvssink
```