awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #560: raspberry pi is not support 720p 1080p???[QUESTION]

MAKER-park, 2020-09-03:
hello !!

 I have confirmed that it works at 640 480 resolution, but when I try to 720 or 1080, I keep getting an error, so I can't try it. Do I need to change the bitrate as well as the resolution?

MAKER-park, 2020-09-07:
oh yeah i try this commandd
gst-launch-1.0 v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480 ! omxh264enc control-rate=2 target-bitrate=512000 periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="MyKinesisVideoStream" access-key="" secret-key="" aws-region="ap-northeast-2"

so i change the resolution 720p 480p but is not working 

and i also try the sample code 
sudo AWS_ACCESS_KEY_ID= AWS_SECRET_ACCESS_KEY= AWS_REGION=ap-northeast-2 ./kvs_gstreamer_sample my-stream-name

but is same resolution 640 480... so how can up resolution?

MushMal, 2020-09-07:
There are many examples of the gst-launch command in  here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md

Please try them and see whether they work for you.

In case it doesn’t work, try to remove kvssink from the pipeline and use fakesink to get your pipeline working. If it fails, please collect the detailed GStreamer logs in debug mode by setting GST_DEBUG=5 environment variable and file an issue in GitHub forums to get some help with the GStreamer pipeline you have.

disa6302, 2020-09-17:
Closing this for now. Feel free to reach out if you have any questions.

closed 2020-09-17 by disa6302

rach21nag, 2022-01-07:
Hello all, 

When I change the resolution- H and W in the below command to 1080p, the video is not streaming. Could you please tell me what changes to make to this code?

gst-launch-1.0 v4l2src device=/dev/video0 \
! videoconvert \
! video/x-raw,format=I420,width=640,height=480 \
! omxh264enc control-rate=2 target-bitrate=512000 periodicity-idr=45 inline-header=FALSE \
! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline \
! kvssink stream-name="raspberry" \