awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #561: [QUESTION] Alignment exception running on 32 bit ARM-Linux

YoungminLouisHan, 2020-09-03:
Dear Amazon,

I've been trying to run the kinesis sdk on ARM 32bit CPU (armv7).
After finishing cross-compile, I've tested it.

Unfortunately, Alignment exception is happened after a few seconds from starting streaming service.
It seems to be related with 32 bit architecture. Because running on Linux Native machine(64 bit) doesn't happen this kind of error.
Could you inform or give any advice to fix this topic? 

Here is my environment :
Kinesis version : v3.0.0 tag
Linux version : 4.14
CPU : armv7 32 bit
Streaming type : video streaming comes from USB camera (uvcvideo module)

A command I've used -
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video1 ! h264parse ! video/x-h264,width=1280,height=720,stream-format=avc,alignment=au ! kvssink stream-name="streaming_name" storage-size=256

[Panic_log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5167632/Panic_log.txt)

MushMal, 2020-09-03:
The default mode is unaligned memory access mode for the content store in PIC. This is done for tight packing. You can change the mode at the compile time.

https://github.com/awslabs/amazon-kinesis-video-streams-pic#note-on-alignment

YoungminLouisHan, 2020-09-07:
@MushMal 
Thanks very much for sharing the information.
It works properly after applying the alignment option.
Let me close this ticket.

closed 2020-09-07 by YoungminLouisHan