awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #563: [FEATURE] Enable key rotation with AWS_WEB_IDENTITY_TOKEN_FILE on gstreamer on Docker

toshitanian, 2020-09-07:
**Is your feature request related to a problem? Please describe.**

In our use case, we run the gstreamer w/ kvs-sdk docker container on EKS. 

I'm using gstreamer with kinesis video stream SDK following [this article](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html). Then it worked fine with authentication using AWS_ACCESS_KEY_ID and AWS_SECRET_ACCESS_KEY.
However it does not work when we run it on EKS with IAM roles for service accounts described [here](https://aws.amazon.com/blogs/opensource/introducing-fine-grained-iam-roles-service-accounts/).
The service accounts gives `AWS_WEB_IDENTITY_TOKEN_FILE ` environment variable to containers to rotate keys.

The issue for same issues about `AWS_WEB_IDENTITY_TOKEN_FILE`  are fixed as below.
aws cli: https://github.com/aws/aws-cli/issues/4477
java sdk: https://github.com/aws/aws-sdk-java-v2/issues/1470

I'm not sure this problem happens just on the docker image (is the code base old?) or current code does not support `AWS_WEB_IDENTITY_TOKEN_FILE`.

I hope gstreamer on docker container works on EKS with IAM roles for service accounts.

**Describe the solution you'd like**

If this sdk used Java sdk backend, upgrade java sdk version.

**Describe alternatives you've considered**

Or update some dependencies...

**Additional context**
Log message is...
```
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: Could not initialize supporting library.
Additional debug info:
/opt/amazon-kinesis-video-streams-producer-sdk-cpp/src/gstreamer/gstkvssink.cpp(1374): gst_kvs_sink_change_state (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:
Failed to init kvs producer. Error: Unable to create Rotating Credential provider. Error status: 0x15000022
ERROR - Unable to create Rotating Credential provider. Error status: 0x15000022
2020-09-07 06:59:18 [140375148595008] Setting pipeline to NULL ...
Freeing pipeline ...
```

toshitanian, 2020-09-07:
The authentication looks rely on https://github.com/awslabs/amazon-kinesis-video-streams-producer-c .

And this issue looks related to https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/99

Please close if you want.

disa6302, 2020-09-17:
@toshitanian ,

We do not support this yet. 

@MushMal might be able to provide more context here?

MushMal, 2020-09-17:
Confirming. Currently, we don't support anything other than 

* Static credential provider
* IoT credential provider
* File-based credential provider

Customers can create their own credential providers based on any one of these as an example.

disa6302, 2020-09-21:
@toshitanian , 

Closing this issue since the question is answered. Feel free to reach out if you have any more questions.

closed 2020-09-21 by disa6302

toshitanian, 2020-09-22:
Thanks a lot.

Hope this supports more flexible key rotation like EC2 iam role and so on.