awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #564: Transfer rate to Kinesis too low

tiensu, 2020-09-10:
I hear that Kinesis Video Stream can work with real-time video.
I tried, but the video stream from my IP camera to Kinesis Video Stream too slow.
This is my command: 
**_gst-launch-1.0 rtspsrc location=rtsp://admin:admin@10.1.34.222:554/ch01/0 short-header=TRUE ! rtph264depay ! kvssink stream-name="stream-d9" storage-size=512 access-key="***" secret-key="***" aws-region="ap-northeast-1"_**
My Internet speed is hight, about 90Mbps upload (test by SpeedTest.net)
Initial transfer_rate is hight (5Mbps) then gradually decrease to around 500Kbps.
Did I make something wrong?

disa6302, 2020-09-17:
@tiensu ,

Updates?

disa6302, 2020-09-21:
@tiensu , 

Closing this issue for now since it has gone stale. Please feel free to reach out if you have any questions or you can reopen this issue as well. Thanks!

closed 2020-09-21 by disa6302