awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #566: vs-buildtools-install.bat not working in MSVC

surekha3079, 2020-09-16:
iam trying to build c++ sdk through url : https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-win.html#producer-sdk-win-msvc
In 4th step iam unable to run this "vs-buildtools-install.bat" file.
Showing error as-- C:\AWS\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>vs-buildtools-install.bat
'vs-buildtools-install.bat' is not recognized as an internal or external command,
operable program or batch file.

closed 2020-09-16 by disa6302

surekha3079, 2020-09-16:
Divya,
Iam new to this AWS.Could u pls provide your mail id for contact. It will be helpful to me

MushMal, 2020-09-16:
@surekha3079 we can't give out personal emails. If you have questions regarding KVS itself, you could cut issues and get some help. If you need to provide or discuss direct matters that require privacy, you could compile an email and send it to kinesis-video-support@amazon.com.

Alternatively, we would encourage you to reach out to your AWS Account Management Team. You can do so from the AWS console.

surekha3079, 2020-09-16:
Ok Thanku.