awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #567: The CXX compiler identification is unknown

surekha3079, 2020-09-17:
Hi, I took latest build .When iam running cmake .. -G "NMake Makefiles" iam facing below error.I installed cmake,pkg-config,gstreamer,VS build tools 2019 also

smanda@VSC-WS216 MINGW64 /c/Users/smanda/amazon-kinesis-video-streams-producer-sdk-cpp-3.0.0/build
$ cmake .. -G "NMake Makefiles"
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:8 (project):

disa6302, 2020-09-17:
@surekha3079 ,

Can you please attach the rest of the logs?

Are you using mingw? If yes, you might have to export CXX

disa6302, 2020-09-17:
This is a sample you could use: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat

Also, check out the windows build in .travis.yml for the steps and see if it works

surekha3079, 2020-09-17:
Actually below is the process iam doing.
1)Download the latest from Release 3.0.0 (14th May 2020) 
By following Readme.mdinstalled `pkg-config` and `CMake` and gstreamer for MINGW64
[image]
2)Running ‘ cmake .. -G "NMake Makefiles" ’  command in build location.Iam getting below error
[image]
But I installed build tools also by selected c++ build tools.
[image]

MushMal, 2020-09-17:
You need to debug this yourself a little. For some reason the paths are not being set properly. Perhaps, add the compiler paths to your %PATH% variable. VS also installs a VS shell that has those paths setup. Check VS support pages and the documentation on how to get that done

surekha3079, 2020-09-18:
Thanku .Now i was able to run by referring **build_windows.bat** file. cmake is able to identify compiler.
But iam facing one issue like..I installed curl and set the path in environment variables also
CMake Error at CMakeLists.txt:106 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.

disa6302, 2020-09-21:
@surekha3079 , 

Based on https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/569, looks like your build is working. Closing this issue. Feel free to reach out if you have any more questions.

closed 2020-09-21 by disa6302