awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #568: How to identify(callback) if any error occurs in GStream while processing?

thasin96, 2020-09-21:
`error_cb` function is being called whenever issue occurs at bus level. For Example, Streaming URL is Not Reachable. It works as I expected but i expect this function will be called even when an issue occurs while processing also. 

i.e., When I run cpp application `kvs_gstreamer_sample.cpp` it checks RTSP url is active or not. Once it's active the applications starts streaming. But in the middle of the process if stream is not reachable error_cb is not being called and I am getting log like below. 

Expected Behaviour: Should call error_cb and send EndOfStream signal.

MushMal, 2020-09-21:
There is a similar issue/enhancement described in issue #375 
This enhancement is specific to handling the GStreamer events in the KVS plugin. We are not planning on enhancing the samples for the readability.

We are more then welcoming any contributions and enhancement to the codebase

disa6302, 2020-09-28:
Closing this.

closed 2020-09-28 by disa6302