awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #569: Could not find a package configuration file provided by "CURL"

surekha3079, 2020-09-21:
Hi I downloaded code from https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.
Trying to build it iam facing below issue..
CMake Error at CMakeLists.txt:106 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "C:/Users/smanda/amazon-kinesis-video-streams-producer-sdk-cpp-3.0.0/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/smanda/amazon-kinesis-video-streams-producer-sdk-cpp-3.0.0/build/CMakeFiles/CMakeError.log".

MushMal, 2020-09-21:
Install CURL library on Windows. Check out their pages and forums for more info

surekha3079, 2020-09-21:
I already downloaded **curl-7.72.0-win64-mingw** and set in environmental variables.  Even though i am facing same

surekha3079, 2020-09-21:
Can u share me the link.. Even i downloaded its showing error

MushMal, 2020-09-21:
I am not really sure why this isn't being picked up. You could try to ping CURL forums to get some help with the installation but I don't see any reasons why it doesn't work.

Meanwhile, on Windows, you could also try VS solution. We have a TravisCI runs with Windows builds you could try to model your install from.

surekha3079, 2020-09-21:
Thanku MushMal..Now i was able to build success...

closed 2020-09-21 by disa6302

aqiao, 2023-08-17:
@surekha3079 , I faced the same issue , could you please share how to fix it?

sirknightj, 2023-08-17:
The issue appears to be that cmake cannot find curl.

The windows build in `develop` is passing. Please check out the `develop` branch and try following the same installation commands with a clean repository (clean build). There's a path variable that gets set there as well on line 295.
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/actions/runs/5882086381/workflow#L278-L297

The build_windows.bat file the ci file refers to can be found [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/4c4afff63f3371a69be2472bb0bee654ef09abd3/.github/build_windows.bat).

aqiao, 2023-08-17:
Switched to develop branch, but show 
```
CMake Error at CMakeLists.txt:119 (find_package):
  Could not find a package configuration file provided by "Log4cplus" with
  any of the following names:

    log4cplusConfig.cmake
    log4cplus-config.cmake

```