awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #570: [QUESTION] Variable Bitrate Support for HLS playback?

suggestedfixes, 2020-09-21:
The problem we are still trying to solve is related to video player trying to playback using different bitrate.

When we experiment with changing bitrate drastically:
Video.js is able to grab the HLS session URL and initialize and update m3u8 playlist.
However, when the player is trying to fetch individual fragments, AWS returns:
Message: "The requested fragment does not have the same codec private data as the rest of the session."

A similar question arises from this observation:
- We were to set our upload stream to variable bitrate, normally although we set at variable bitrate, the actual bitrate will not vary too much. But there will be cases where the bitrate will change due to activity, thus might emulate the problem we are having with manually changing the bitrate.
- However, Would we better off setting constant bitrate?

MushMal, 2020-09-21:
@suggestedfixes you can think of it whether the CPD has changed or not. Most of the time, the ABR change will not result in an H264 SPS/PPS changes and the streaming can go uninterrupted. However, currently, KVS HLS playback end-to-end is requiring the same CPD. 

There is a technical difficulty with streaming/indexing ABR streams and we are considering in the future to provide a proper support - at least when the underlying stream is in Annex-B format.

I think for now you have two options.

1) Use CBR streaming
2) Use different sessions if you need to bitrate switch in the encoder. You will need to stop stream/free stream, re-create a new stream with new CPD. This is going to end up having glitches in the end stream, not to mention that you will need to implement your own HLS based on ListFragments/GetMediaForFragment APIs.

disa6302, 2020-09-28:
Closing this based on tag. @suggestedfixes , feel free to reach out if you have any more questions.

closed 2020-09-28 by disa6302