awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #572: [QUESTION] 0x52000062 error

staowyze, 2020-09-22:
Hi Experts,

We got following errors, would it cause re-connecting in the SDK, and try to get data endpoint thereafter?

`Upload handle 14. Fragment timecode in 100ns: 16008147175540000. Error status: 0x52000062`

Thank you!

MushMal, 2020-09-22:
This is 0x52000062 | STATUS_ACK_ERR_INVALID_MKV_DATA | Invalid MKV data as defined in https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html

Any backend error means the backend will terminate the connection so yes, it will evaluate to termination.

This is likely caused by the partially submitted frame being dropped on the buffer pressure. We have a fix in the later versions of the SDK that will attempt to not drop the partial frame which would still cause a frame drop on the pressure but will avoid the syntactic error causing this.

Recommend updating to latest to minimize the chance of getting INVALID_MKV. The SDK will attempt to reset the connection on the buffer pressure which evaluates to another PutMedia call but this is done in hopes that the new connection will be faster and will allow clearing the buffer.

staowyze, 2020-09-23:
@MushMal thank you for the answer!
One more question, we registered error callback at client side, and didn't see errors, however cloudwatch still showed PutMedia.ConnectionErrors, a couple of questions
1. what's connectionerrors exactly?
2. is it possible to query what are the connection errors?
3. we see much lower PutMedia.Requests after some fixes, what's PutMedia.Requests exactly?

Thank you!

MushMal, 2020-09-23:
There are different types of errors. For this conversation we should focus on the runtime or dynamic errors that happen during the transmission. Those fall into two categories.

1) Errors that are related to the actual streamed data and we have a graceful reporting of an Error ACK from the backend following by a connection closure.
2) Not graceful errors which simply terminate the connection - these can be due to the ingestion host death for example or LB dropping the connection or anything else that can terminate the connection.

In the first case, we do report an error via the error callback and you could also subscribe to the fragment ACK received callback and listen to the error ACK type.

In the latter case, it will depend - most of the errors will reach the SDK in a form of terminated connection or connection staleness, etc... The SDK will automatically recover (default configuration) so the errors will not be reported.

Connection errors reported in the CW is the collective count of all of the errors - including the transient ones like the termination of the connection. This CW metric does not show the actual error - tracking those will be cost prohibitive and we only do deep dive into the logs per need.

PutMedia requests means how many streaming sessions there is. For example, if your device is streaming continuously and no interruptions happen then you will have a single putmedia request. However, any time there is token rotation or a disconnect, the SDK will retry to connect and send - each streaming session is a single PutMedia.

Please resolve the issue if you have your questions answered. I would encourage you to cut a new issue if the question is unrelated to the original question posted in this issue.

closed 2020-09-23 by staowyze