awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #573: JSMN is a build time dependency, but not named in documentation

rpcme, 2020-09-23:
**Describe the bug**
During `cmake -DBUILD_DEPENDENCIES=OFF` configure fails, even though all named dependencies in documentation have been configured.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone fresh, recursively, tag 3.0.0
2. Invoke `cmake -DBUILD_DEPENDENCIES=OFF ..`, or using whatever CMAKE_BUILD and CMAKE_SOURCE directories you desire
3. Witness the following error

```bash
...
)
| -- Could NOT find JSMN (missing: JSMN_LIBRARY JSMN_INCLUDE_DIRS)
| -- Bitness 64 bits
| -- Check if compiler accepts -pthread
| -- Check if compiler accepts -pthread - yes
| CMake Error at CMakeLists.txt:127 (find_library):
|   Could not find Jsmn using the following names:
| 
| 
| -- Configuring incomplete, errors occurred!
...
```

**Expected behavior**
Configure occurs normally with documented dependencies implemented

**SDK version number**

3.0.0

**Open source building**

This library is not defined in dependencies

**Screenshots**
Not applicable

**Platform (please complete the following information):**
 - OS: Ubuntu
 - Processor: x86_64
 - Version 18.04

**Additional context**
Cross compilation to qemu x86_64

disa6302, 2020-09-28:
@rpcme , 

Apologies for not including in the documentation. Is jsmn available in the search path? 
Which path contains the jsmn? Maybe try to add a search path to point to that?

MushMal, 2020-10-10:
The CMake scripts should properly install this dependency.

closed 2020-10-10 by MushMal