awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #575: [QUESTION] PutMedia.ConnectionErrors occurs when no data is uploaded

staowyze, 2020-09-24:
Hi experts,

From cloudwatch, we see persistent PutMedia.ConnectionErrors when we stop sending data to KVS. We have the following client side log
```
[09-24 18:36:07] curlCompleteSync(): curl perform failed for url https://s-xxxxxxx.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[09-24 18:36:07] curlCompleteSync(): HTTP Error 0 : Response: (null)
[09-24 18:36:07] getStreamingEndpointCachingCurl(): getStreamingEndpointCachingCurl
```

Is the connection error caused by putMedia timeout?

staowyze, 2020-09-24:
@MushMal may i have your attention on this? this is a critical issue we want to confirm.

Thank you!

MushMal, 2020-09-24:
It is a timeout issue . Without the logs and more info I  am not sure what’s causing it - perhaps simply the fact that you stop producing frame s and are not using EoFR

staowyze, 2020-09-24:
what level of logs is needed? we are using 2.0.1, the log level is warn.

MushMal, 2020-09-24:
You might want to enable debug level logging and log for longer period of time to ensure this is what you are seeing - namely, the producer timing out due to no frames being pushed into the SDK. The older versions of the SDK would keep retrying the session and would keep timing out. You could update to the latter version of the SDK or try to close the stream at the end of the streaming session. Then, with the new one you would re-create the stream object or simply reset the stream in order to stream a new session. 

Note: recreating a stream will evaluate to going through describe and get data endpoint states before issuing a new PutMedia call

staowyze, 2020-09-25:
Would this good enough?
```
[09-24 23:58:26] createKinesisVideoClient(): Creating Kinesis Video Client
[09-24 23:58:26] heapInitialize(): Initializing native heap with limit size 9437184, spill ratio 0% and flags 0x00000001
[09-24 23:58:26] heapInitialize(): Creating AIV heap.
[09-24 23:58:26] heapInitialize(): Heap is initialized OK
[09-24 23:58:26] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x1
[09-24 23:58:26] stepStateMachine(): PIC Client State Machine - Current state: 0x38e0f5, Next state: 0x2
[09-24 23:58:26] createDeviceResultEvent(): Create device result event.
[09-24 23:58:26] stepStateMachine(): PIC Client State Machine - Current state: 0x12, Next state: 0x10
[09-25 00:45:22] createKinesisVideoStream(): Creating Kinesis Video Stream.
[09-25 00:45:22] stepStateMachine(): PIC Stream State Machine - Current state: 0x0, Next state: 0x1
[09-25 00:45:22] stepStateMachine(): PIC Client State Machine - Current state: 0x8762b8, Next state: 0x40
[09-25 00:45:23] convertTimestampToEpoch(): Expiration timestamp conversion into tm structure  120-8-25T1:45:24
[09-25 00:45:23] convertTimestampToEpoch(): Difference between current time and iot expiration is 3601 
[09-25 00:45:23] parseIotResponse(): Iot credential expiration time 1600998324
[09-25 00:45:23] stepStateMachine(): PIC Client State Machine - Current state: 0x38e0fb, Next state: 0x2
[09-25 00:45:23] describeStreamCachingCurl(): No-op DescribeStream API call
[09-25 00:45:23] describeStreamResultEvent(): Describe stream result event.
[09-25 00:45:23] stepStateMachine(): PIC Stream State Machine - Current state: 0x402490, Next state: 0x2
[09-25 00:45:23] stepStateMachine(): PIC Client State Machine - Current state: 0x76c6035c, Next state: 0x40
[09-25 00:45:23] getKinesisVideoMetrics(): Get the memory metrics size.
[09-25 00:45:23] logStreamMetric(): Kinesis Video client and stream metrics: 
[09-25 00:45:24] getKinesisVideoMetrics(): Get the memory metrics size.
[09-25 00:45:24] logStreamMetric(): Kinesis Video client and stream metrics: 
[09-25 00:45:24] writeHeaderCallback(): RequestId: 64216e4a-dd8a-481b-936b-83c4126ad139
[09-25 00:45:24] postResponseWriteCallback(): postResponseWriteCallback (curl callback) invoked
[09-25 00:45:24] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-xxxxxxxxx.kinesisvideo.us-west-2.amazonaws.com"}
[09-25 00:45:24] terminateCurlSession(): Force stopping the curl connection
[09-25 00:45:24] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[09-25 00:45:24] stepStateMachine(): PIC Stream State Machine - Current state: 0x76d2ca7c, Next state: 0x20
[09-25 00:45:24] stepStateMachine(): PIC Client State Machine - Current state: 0x76d51484, Next state: 0x40
[09-25 00:45:24] getStreamingTokenResultEvent(): Get streaming token result event.
[09-25 00:45:24] stepStateMachine(): PIC Stream State Machine - Current state: 0x750b2360, Next state: 0x10
[09-25 00:45:24] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 9068541
[09-25 00:45:24] stepStateMachine(): PIC Stream State Machine - Current state: 0x76d0cc2c, Next state: 0x40
[09-25 00:45:24] stepStateMachine(): PIC Client State Machine - Current state: 0x0, Next state: 0x40
[09-25 00:45:24] putStreamResultEvent(): Put stream result event.
[09-25 00:45:24] notifyDataAvailable(): Note data received: duration(100ns): 31970000 bytes 331902 for stream handle 0
[09-25 00:45:24] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 31970000. Bytes available: 331902
[09-25 00:45:24] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 31970000. Bytes available: 331902
[09-25 00:45:24] stepStateMachine(): PIC Stream State Machine - Current state: 0x76cdf5b0, Next state: 0x80
[09-25 00:45:24] notifyDataAvailable(): Note data received: duration(100ns): 31990000 bytes 334320 for stream handle 0
[09-25 00:45:24] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 31990000. Bytes available: 334320
[09-25 00:45:24] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 31990000. Bytes available: 334320
[09-25 00:45:24] notifyDataAvailable(): Note data received: duration(100ns): 32440000 bytes 334653 for stream handle 0
[09-25 00:45:24] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 32440000. Bytes available: 334653
[09-25 00:45:24] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 32440000. Bytes available: 334653
[09-25 00:45:24] notifyDataAvailable(): Note data received: duration(100ns): 32810000 bytes 334986 for stream handle 0
[09-25 00:45:24] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 32810000. Bytes available: 334986
[... cut ...]
[09-25 00:45:27] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 23040000. Bytes available: 276788
[09-25 00:45:27] postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
[09-25 00:45:27] postReadCallback(): Get Stream data returned: buffer size: 16372 written bytes: 16372 for upload handle: 0 current stream handle: 9068541
[09-25 00:45:27] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[09-25 00:45:27] postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
```

staowyze, 2020-09-25:
more logs with connection errors

```
[09-25 00:46:09] postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
[09-25 00:46:09] postReadCallback(): Get Stream data returned: buffer size: 16372 written bytes: 15947 for upload handle: 0 current stream handle: 9068541
[09-25 00:46:09] postReadCallback(): Wrote 15947 bytes to Kinesis Video. Upload stream handle: 0
[09-25 00:46:09] postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
[09-25 00:46:09] postReadCallback(): Get Stream data returned: buffer size: 16372 written bytes: 0 for upload handle: 0 current stream handle: 9068541
[09-25 00:46:09] postReadCallback(): Pausing CURL read for upload handle: 0
[09-25 00:46:09] notifyDataAvailable(): Note data received: duration(100ns): 0 bytes 333 for stream handle 0
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 0. Bytes available: 333
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 0. Bytes available: 333
[09-25 00:46:09] notifyDataAvailable(): Note data received: duration(100ns): 430000 bytes 666 for stream handle 0
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 430000. Bytes available: 666
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 430000. Bytes available: 666
[09-25 00:46:09] notifyDataAvailable(): Note data received: duration(100ns): 600000 bytes 5073 for stream handle 0
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 600000. Bytes available: 5073
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 600000. Bytes available: 5073
[09-25 00:46:09] notifyDataAvailable(): Note data received: duration(100ns): 800000 bytes 5406 for stream handle 0
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 800000. Bytes available: 5406
[09-25 00:46:09] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 800000. Bytes available: 5406
[09-25 00:46:09] notifyDataAvailable(): Note data received: duration(100ns): 1200000 bytes 5739 for stream handle 0
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1200000. Bytes available: 5739
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1200000. Bytes available: 5739
[09-25 00:46:10] notifyDataAvailable(): Note data received: duration(100ns): 1210000 bytes 9387 for stream handle 0
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1210000. Bytes available: 9387
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1210000. Bytes available: 9387
[09-25 00:46:10] notifyDataAvailable(): Note data received: duration(100ns): 1750000 bytes 9720 for stream handle 0
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1750000. Bytes available: 9720
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1750000. Bytes available: 9720
[09-25 00:46:10] notifyDataAvailable(): Note data received: duration(100ns): 1940000 bytes 13545 for stream handle 0
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1940000. Bytes available: 13545
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 1940000. Bytes available: 13545
[09-25 00:46:10] notifyDataAvailable(): Note data received: duration(100ns): 2010000 bytes 13878 for stream handle 0
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 2010000. Bytes available: 13878
[09-25 00:46:10] defaultStreamDataAvailableCallback(): Reported streamDataAvailable callback for stream handle 9068541. Upload handle 0. Duration available in 100ns: 2010000. Bytes available: 13878
[09-25 00:46:10] notifyDataAvailable(): Note data received: duration(100ns): 2410000 bytes 14211 for stream handle 0
[... cut ...]
[09-25 00:53:02] postReadCallback(): Get Stream data returned: buffer size: 16372 written bytes: 14433 for upload handle: 11 current stream handle: 9068541
[09-25 00:53:02] postReadCallback(): Wrote 14433 bytes to Kinesis Video. Upload stream handle: 11
[09-25 00:53:02] postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
[09-25 00:53:02] postReadCallback(): Get Stream data returned: buffer size: 16372 written bytes: 0 for upload handle: 11 current stream handle: 9068541
[09-25 00:53:02] postReadCallback(): Pausing CURL read for upload handle: 11
```

MushMal, 2020-09-25:
Confirming. From the log indeed it shows that your applications stops/pauses producing frames causing the timeout. My suggestion above is applicable - best is to update the SDK. Alternatively, you could recreate the stream with v2.x when a new “session” starts

staowyze, 2020-09-25:
thank you!
We don't want to recreate the stream when a new event starts, since recreate the stream would clear endpoint cache, and cause more get data endpoint API calls. We are close to the API TPS limit.

staowyze, 2020-09-25:
is there a way we could achieve the same goal without clearing endpoint cache?

MushMal, 2020-09-25:
Updating to the latest SDK will allow automatic handling of the intermittent producer with caching endpoint. Using older sdk you could stopSync the stream and reset the stream before streaming with caching APIs. It should work fine

staowyze, 2020-09-25:
Good to know, thank you!

closed 2020-09-25 by staowyze