awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #576: [QUESTION] Content of Dockerfile used in AWS documentation.

toshitanian, 2020-09-28:
We're using Docker image described in [AWS documentation](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker-download). Specifically `546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux:latest`.
Now we're facing a memory leak problem and want to look into more details. So I need to re-build docker image from Dockerfile. But I'm not able to find Dockerfile in this repo.

I found a [sample repository](https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/producer-cpp) but I'm not sure this is the Dockerfile used in AWS documentation.

So in terms of docker image in AWS documentation.
- (1) If you have your internal Dockerfile, let us use it by adding it to this repo.
- (2) If the docker image is built from sample repo, we should add the description to README

Thanks,

disa6302, 2020-09-28:
@toshitanian , 
546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux:latest should contain the latest. And yes, the docker in the sample repository is the one that is being used.

About the README.md, you are right. Maybe we lost the link in some commit. Let me add a PR to link to the demos repo.

toshitanian, 2020-09-28:
Thanks a lot !

PR #577 (2020-11-17): Add dockerscript link to README.md
*Issue #, if available:*
#576 
*Description of changes:*

Resolves #576 
By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2020-10-10 by MushMal