awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #579: [QUESTION] Unable to compile the code for building GStreamer plugin

HarkiratGrewal, 2020-09-29:
Just to mention, I have littke to no knowledge of c++ or working of it's compiler.
I have been trying to build the plugin for GStreamer on my local machine:
OS: Windows 10 Home
Here are the steps, I followed:
1. Cloning the github: 
```
git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
```
2. Run the follwing command as Admin: git config --system core.longpaths true
3. Built the directory and navigated to it: 
```
mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
cd amazon-kinesis-video-streams-producer-sdk-cpp/build
```
4. Run the following command for starting the build in Developer's command promt for VS 16 2019: 
```
cmake .. -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON
```
Here is the error, I am getting:
```
        lib /nologo /out:apps\libapps.lib @C:\Users\<username>\AppData\Local\Temp\nm38C6.tmp
LINK : fatal error LNK1181: cannot open input file 'apps\app_rand.obj'
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\lib.EXE"' : return code '0x49d'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\Hostx86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: 'echo' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:64 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "C:/SDK/C++/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
```
I guess the issue is with the OpenSSL library itself. Is it possible if instead of latest version for OpenSSL, we could get a previous version that was working. This could take a lot of time to debug but maybe a workaround would be that of releasing a version for GStreamer plugin.

MushMal, 2020-10-10:
Please try the latest commit which should fix the versioning issue

closed 2020-10-10 by MushMal

pestisid-cmd, 2024-03-04:
Still facing the same issue when trying to build CPP SDK.