awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #580: [QUESTION] More Lenient Requirements for Bitrate change

suggestedfixes, 2020-09-30:
This is a follow up to this issue:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/570

In summary, if we change the camera bitrate and while uploading to KVS, we get inconsistent codec private data error during playback if the session window crosses the time where we changed the bitrate.

We did more experiments, the variables of concern are:
- variable bitrate mode or constant bitrate mode.
- actual bitrate.

What we found:
- changing between CBR/VBR does not change the CPD.
- changing bitrate changes CPD.
- software/hardware restart does not change the CPD.

Our question is, since the only change is the bitrate change. How difficult would it for KVS to be more lenient on the bitrate change requirement?

Sample CPDs, h264/high/20fps/20gop/1600x1200:

256 kbps, CBR/VBR
01640028ffe1002e67640028ac1b1a8064025ffc05b808080a000007d000013881d0c07400002faf05de5c68603a000017d782ef2e1401000468ee3830

512 kbps, CBR/VBR
01640028ffe1002f67640028ac1b1a8064025ffc05b808080a000007d000013881d0c03a00000bebc177971a18074000017d782ef2e14001000468ee3830

1024 kbps, CBR/VBR
01640028ffe1002f67640028ac1b1a8064025ffc05b808080a000007d000013881d0c01d080002faf05de5c68600e8400017d782ef2e1401000468ee3830

suggestedfixes, 2020-09-30:
Another interesting phenomenon we observed after some more testing:
If we were to change the bitrate only on the camera's web interface.
Observations:
-	Camera does not reboot
-	Our app does not supply a new CPD.
-	Producer still able to upload video with our incorrect CPD.
-	The stream is still  downloadable.
-	The browser side does not complain with wrong CPD.

MushMal, 2020-09-30:
@suggestedfixes perhaps I am not sure what the question is. I will try to provide some answers as I know.

* VBR vs CBR. There might not be an immediate change in CPD between these when it comes to generating the CPD. Just need to ensure that the VBR implementation doesn't actually end up doing ABR-type bit shifts where it could generate a new CPD. Also, make sure your downstream consumers can consume VBR - most don't. Most of the VBR implementations are actually cVBR (Capped VBR).
* Yes, changing the bitrate might imply codec quality selection changes. Same with resolution changes and any other algorithmic selections on the encoder side. This is highly encoder dependent.
* SW vs HW encoders. Again, this is purely dependent on the encoder settings. If the settings are identical then the CPDs will be logically identical. Might have some packaging differences.
* There is nothing KVS can do when it comes to CPDs. First, changing the bitrate does not directly imply changes in CPD - it is likely implying the quality or compression ratio changes in the encoder settings, which, in turn affects the CPD. The KVS main backend can ingest anything and egress anything you put into it. It's impacting higher-level concepts such as HLS/MPEG-DASH or console playback that need the immediate knowledge of the CPD. To give a little more background. If your solution does not depend on the HLS/MPEG-DASH playback on KVS but rather you implement your own, then your producer can omit the CPD all together and you could communicate the CPD changes offline or embed it in the stream. For example, h264 in an Annex-B format has many implementations on the encoder side where the SPS and the PPS are pre-pended to the I-frame. In this way, the resolution or the bitrate could change dynamically. As long as the downstream dependencies consume this. Most likely, your producer uses Annex-B format which allows the browser playback to auto-stitch the CPD.

suggestedfixes, 2020-10-01:
The current symptom is video.js can download the playlist, but when video.js tries to fetch individual fragments, it's the individual fragment URL that complains about inconsistencies of CPD.

suggestedfixes, 2020-10-01:
So I'm asking if it's possible to not to generate an error message, but instead, let the player download the fragments and play them.

MushMal, 2020-10-01:
You can't use KVS HLS or MPEG-DASH (or other high-level Reader) APIs with fragments of different CPD. You will have to implement your own.  As is, KVS does not support CPD switching in mid-flight in Reader APIs

MushMal, 2020-10-01:
I would like to get @MixMasterMitch to comment on the feasibility but as is, there is no way to achieve it

suggestedfixes, 2020-10-01:
So you would suggest using get media for fragment list API?
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_GetMediaForFragmentList.html

MushMal, 2020-10-01:
I really don’t want to be in a position to suggest your architectural solution but I could give a common directions. Think about whether you need to do any CPD switching in the first place and whether you will be saving much. Think if you want to have dual stream low and high bitrate per device which could start stop based on the network pressures, etc. unfortunately you won’t have out of the box solution

suggestedfixes, 2020-10-01:
So the dual-stream would contain 2 tracks, correct? And HLS session would have the metadata for stream selection?

suggestedfixes, 2020-10-01:
Or do you mean streaming into separate channels?

MushMal, 2020-10-01:
Well, again, I would like to understand what your application is and what it's trying to do. Are you trying to do ABR due to upload network bandwidth issues? Do the network bottlenecks have temporary character? What's your latency requirements? What's your consumer story?... Many questions need to be answered before moving forward. 

For example, in some cases, when the Producer side needs to handle temporary network outage or bandwidth limitation, it could simply define a larger buffer which could hold minutes or hours long video. Some of the customers have solutions based on simply stopping the upload and issuing an alarm if the network issue keeps happening over a long period of time.

The solution with dual stream is about having two or even more separate streams created - each for a given bitrate. When the network is having an issue, the producer application stops the high-bitrate stream and creates a stream object with the lower bitrate stream. The consumer will need to be aware of the logic and issue a manifest refresh to stitch the sessions from different bitrate streams.

Yet, if you can implement your own playback, you could simply have an ABR stream with the CPD embedded in Annex-B format I-frames that can ABR with every fragment without the need for a higher-level support such as HLS or MPEG-DASH.

suggestedfixes, 2020-10-01:
Our consumer story is that the consumer can adjust the camera settings at any time, but once they do that. There is a discontinuity in the historical stream (fragment duration * size of the playlist) where the footage is no longer streamable because of inconsistent codec data.

suggestedfixes, 2020-10-01:
I'm really thinking outside of the box here. Gathering all of the things you provided, is it possible if we supply a consistent dummy codec private data which looks like a generic h264 with VBR with max bitrate, and the player will be able to decode that as long as each fragment contains an I-frame with SPS and PPS prepended?

MushMal, 2020-10-01:
Actually, that's a very compelling story. Here are some ideas you could play with

1) Producer has a way of notifying the consumer about the times when the CPD has changed. Thus, the consumer would issue a different playback sessions, each will be continuous in itself with a certain CPD. Most players would be fast to switch the HLS sessions. There are pros and cons to this approach.
2) Depending on the player, if you have access to the player you could inspect the CPD yourself and see if the new fragment is any different and if so, you could potentially create a new session. The con here is that you need to have a logic on the download path.
3) Have a custom player based on GetMedia or GetMediaForFragment API. In this case, you could simply have a playback based on your player submitting the CPD whenever it changes. 

Your idea about dummy CPD is an interesting one. Honestly, I need to look into it but give it a try and see how your player reacts.

MixMasterMitch, 2020-10-01:
KVS already has a feature request open to support codec switching by using a discontinuity flag. No ETA on that.

As @MushMal said, dummy CPD could be a workaround for you.

closed 2020-10-10 by MushMal