awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #581: kvs_gstreamer_sample.cpp application throws `Not all elements could be created`

thasin96, 2020-10-05:
I was using kvs_gstreamer_multistream_sample.cpp application to stream but now I switched to kvs_gstreamer_sample.cpp application. This throws `Not all elements could be created`for the same camera. It was working properly for multistream application.

Complete Error Log:

```
[INFO ] [01-10-2020 13:33:54:116.723 GMT] Using region: ap-southeast-1
[INFO ] [01-10-2020 13:33:54:117.366 GMT] Using IoT credentials for Kinesis Video Streams
[DEBUG] [01-10-2020 13:33:54:123.404 GMT] Creating IoT auth callbacks.
[INFO ] [01-10-2020 13:33:54:337.951 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [01-10-2020 13:33:54:338.541 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [01-10-2020 13:33:54:338.918 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [01-10-2020 13:33:54:340.897 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [01-10-2020 13:33:54:341.458 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [01-10-2020 13:33:54:341.905 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [01-10-2020 13:33:54:342.296 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [01-10-2020 13:33:54:342.726 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [01-10-2020 13:33:54:343.146 GMT] clientReadyHandler invoked
[DEBUG] [01-10-2020 13:33:54:343.858 GMT] Client is ready
[INFO ] [01-10-2020 13:33:54:344.874 GMT] Creating Kinesis Video Stream stream_3bb896f0-53a9-4458-9a85-4c2b96192495_14127b5b-03d8-4bd1-ad6f-f3c06d5ce2de
[INFO ] [01-10-2020 13:33:54:345.588 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [01-10-2020 13:33:54:345.997 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [01-10-2020 13:33:54:346.405 GMT] logStreamInfo():      Stream name: stream_3bb896f0-53a9-4458-9a85-4c2b96192495_14127b5b-03d8-4bd1-ad6f-f3c06d5ce2de        
[DEBUG] [01-10-2020 13:33:54:346.678 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [01-10-2020 13:33:54:346.885 GMT] logStreamInfo():      Content type: video/h264
[DEBUG] [01-10-2020 13:33:54:347.118 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [01-10-2020 13:33:54:347.326 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [01-10-2020 13:33:54:347.527 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [01-10-2020 13:33:54:347.825 GMT] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [01-10-2020 13:33:54:348.127 GMT] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [01-10-2020 13:33:54:348.393 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [01-10-2020 13:33:54:348.644 GMT] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [01-10-2020 13:33:54:348.900 GMT] logStreamInfo():      Framerate: 25
[DEBUG] [01-10-2020 13:33:54:349.204 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [01-10-2020 13:33:54:349.535 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [01-10-2020 13:33:54:349.755 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [01-10-2020 13:33:54:349.996 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [01-10-2020 13:33:54:350.246 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [01-10-2020 13:33:54:350.455 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [01-10-2020 13:33:54:350.669 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [01-10-2020 13:33:54:350.905 GMT] logStreamInfo(): Track list
[DEBUG] [01-10-2020 13:33:54:351.122 GMT] logStreamInfo():      Track id: 1
[DEBUG] [01-10-2020 13:33:54:351.365 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [01-10-2020 13:33:54:351.577 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [01-10-2020 13:33:54:351.818 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [01-10-2020 13:33:54:352.116 GMT] logStreamInfo():      Track cpd: NULL
[DEBUG] [01-10-2020 13:33:54:352.984 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [01-10-2020 13:33:54:506.228 GMT] writeHeaderCallback(): RequestId: b88af9a2-8854-494b-a01d-ca9b55bcc8a2
[DEBUG] [01-10-2020 13:33:54:507.877 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.6015568729E9,"DataRetentionInHours":24,"DeviceName":"3bb896f0-53a9-4458-9a85-4c2b96192495_14127b5b-03d8-4bd1-ad6f-f3c06d5ce2de","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-southeast-1:709356462602:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-southeast-1:7093562602:stream/stream_3bb896f0-53a9-4458-9a85-4c2b96192495_14127b5b-03d8-4bd1-ad6f-f3c06d
[INFO ] [01-10-2020 13:33:54:520.543 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [01-10-2020 13:33:54:520.924 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [01-10-2020 13:33:54:521.205 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [01-10-2020 13:33:54:662.532 GMT] writeHeaderCallback(): RequestId: 3d99c95f-7eb1-4e36-98d2-cd695d8c53e7
[DEBUG] [01-10-2020 13:33:54:665.734 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://fgfgdrg.kinesisvideo.ap-southeast-1.amazonaws.com"}
[INFO ] [01-10-2020 13:33:54:680.418 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [01-10-2020 13:33:54:680.982 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[INFO ] [01-10-2020 13:33:54:681.472 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [01-10-2020 13:33:54:681.929 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [01-10-2020 13:33:54:682.364 GMT] streamReadyHandler invoked
[DEBUG] [01-10-2020 13:33:54:683.327 GMT] Stream is ready
[INFO ] [01-10-2020 13:33:54:730.731 GMT] Streaming from rtsp source
Not all elements could be created.
[INFO ] [01-10-2020 13:33:54:761.873 GMT] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 000000557615fac8.
[DEBUG] [01-10-2020 13:33:54:762.257 GMT] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ] [01-10-2020 13:33:54:762.533 GMT] Freeing Kinesis Video Stream stream_3bb896f0-53a9-4458-9a85-4c2b96192495_14127b5b-03d8-4bd1-ad6f-f3c06d5ce2de
[INFO ] [01-10-2020 13:33:54:762.767 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG] [01-10-2020 13:33:54:762.953 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[INFO ] [01-10-2020 13:33:55:165.391 GMT] Freeing Kinesis Video Stream stream_3bb896f0-53a9-4458-9a85-4c2b96192495_14127b5b-03d8-4bd1-ad6f-f3c06d5ce2de
[INFO ] [01-10-2020 13:33:55:166.837 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [01-10-2020 13:33:55:167.645 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [01-10-2020 13:33:55:569.436 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [01-10-2020 13:33:55:570.570 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

```

How to resolve this error?

hassanctech, 2020-10-05:
It looks like we print out that exact same error message in two different places.  Could you please go here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/76e697c13e61de4fda29e0ec050339532fdfb8ea/samples/kvs_gstreamer_sample.cpp#L781

And modify the message to be:  "Failed to create videoconvert element."  Then please run it again so we can narrow down where exactly it's failing.

The kvs_gstreamer_sample loads several more plugins than kvs_gstreamer_multistream_sample, so we just need to figure out which one you're missing.

If after another run it prints that it failed to create videoconvert element then you'll just need to get that plugin.  Alternatively if it still prints the other message then we'll have to go through the elements here:  

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/76e697c13e61de4fda29e0ec050339532fdfb8ea/samples/kvs_gstreamer_sample.cpp#L693-L712

As a quick test immediately before Line 714 you can have a series of if's checking if each individual plugin is null and if it is, print out which one, like this (I didn't include all of them because some of them overlap with kvs_gstreamer_multistream_sample, which you said is already working for you):

`
if (!source) {
  g_printerr("Source element is NULL\n");
} 

if (!encoder) {
  g_printerr("Encoder element is NULL\n");
}

if (!h264parse) {
  g_printerr("h264parse element is NULL\n");
}
`

thasin96, 2020-10-05:
Thanks for the response. I figured out the issue. Some lib packages were missed. After installing gstreamer-1.0bad the issue was resolved.

closed 2020-10-05 by thasin96