awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #582: [BUG] Unit size of storage-size parameter for gstreamer is different by code and documents

toshitanian, 2020-10-06:
**Describe the bug**

In the [README](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#using-element
) of this repo and [AWS documentation](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin-parameters.html), it is describe that the unit of storage-size parameter of gstreamer should be kilobytes.

On the other hand, In the code base, the unit of storage size looks intended to be megabytes.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/76e697c13e61de4fda29e0ec050339532fdfb8ea/src/gstreamer/gstkvssink.cpp#L538-L541

Is my understanding wrong?
If there's difference, It's better to fix the document.

toshitanian, 2020-10-06:
It worked for me by specifying with Mbytes

MushMal, 2020-10-06:
I would have preferred this to be in kilobytes but we won't change this to KB in the code. Instead, the documentation will need to change. Would you like to push a PR for the README fix? Please let me know. Alternatively, I will update the doc when I make some changes later

toshitanian, 2020-10-06:
I will make PR and let you know. Thanks

hassanctech, 2020-11-02:
Merged.  Thank you @toshitanian

closed 2020-11-02 by hassanctech