awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #585: kinesis-video-c-producer get error status 0x52000085 after running for a period of time

laoyouji007, 2020-10-08:
Hello, when I sent video and audio frames using kinesis-video-c-producer, after running for a period of time, I get some error status of 0x52000085. What's the meaning of 0x52000085, it means that the video frames is much more than audio frames? Or is the time interval between two video frames or two audio frames too long?

Here are the logs:
2020-10-07 13:35:48 heapInitialize(): Initializing native heap with limit size 2097152, spill ratio 0% and flags 0x00000001
2020-10-07 13:35:48 heapInitialize(): Creating AIV heap.
2020-10-07 13:35:48 heapInitialize(): Heap is initialized OK
2020-10-07 13:35:48 stepStateMachine(): PIC Client State Machine - Current state: 0x0, Next state: 0x1
2020-10-07 13:35:48 stepStateMachine(): PIC Client State Machine - Current state: 0x0, Next state: 0x2
2020-10-07 13:35:48 createDeviceResultEvent(): Create device result event.
2020-10-07 13:35:48 stepStateMachine(): PIC Client State Machine - Current state: 0x0, Next state: 0x10
2020-10-07 13:35:48 createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-10-07 13:35:48 stepStateMachine(): PIC Stream State Machine - Current state: 0x1756810c, Next state: 0x1
2020-10-07 13:35:48 stepStateMachine(): PIC Client State Machine - Current state: 0xb6ff6dec, Next state: 0x40
2020-10-07 13:35:49 writeHeaderCallback(): RequestId: 5589a72f-4bbf-4522-a00b-d8f0a4a218e7
2020-10-07 13:35:49 describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.602036620978E9,"DataRetentionInHo
urs":2,"DeviceName":"JY9757ACX1EVE7NM","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-2:428194500564:alias/aws/kinesisvideo","Me
diaType":"video/h264,audio/aac","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-2:428194500564:stream/6023A4BC43AC/1602036620978","
StreamName":"6023A4BC43AC","Version":"aBx3WiL1NSfF2SZzTwFe"}}
2020-10-07 13:35:49 describeStreamResultEvent(): Describe stream result event.
2020-10-07 13:35:49 stepStateMachine(): PIC Stream State Machine - Current state: 0xb66d6390, Next state: 0x2
2020-10-07 13:35:49 stepStateMachine(): PIC Client State Machine - Current state: 0xb66d6390, Next state: 0x40
2020-10-07 13:35:51 writeHeaderCallback(): RequestId: b2450c78-2501-4a1a-b3c2-60a850451bea
2020-10-07 13:35:51 getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-171c51c7.kinesisvideo.us-e
ast-2.amazonaws.com"}
2020-10-07 13:35:51 getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2020-10-07 13:35:51 stepStateMachine(): PIC Stream State Machine - Current state: 0x0, Next state: 0x20
2020-10-07 13:35:51 stepStateMachine(): PIC Client State Machine - Current state: 0x1756810c, Next state: 0x40
2020-10-07 13:35:51 getStreamingTokenResultEvent(): Get streaming token result event.
2020-10-07 13:35:51 stepStateMachine(): PIC Stream State Machine - Current state: 0xf0, Next state: 0x10
2020-10-07 13:35:51 defaultStreamReadyCallback(): Reported streamReady callback for stream handle 2371213
2020-10-07 13:35:51 stepStateMachine(): PIC Stream State Machine - Current state: 0x0, Next state: 0x40
2020-10-07 13:35:51 stepStateMachine(): PIC Client State Machine - Current state: 0x1756810c, Next state: 0x40
2020-10-07 13:35:51 putStreamResultEvent(): Put stream result event.
2020-10-07 13:35:51 stepStateMachine(): PIC Stream State Machine - Current state: 0xb66d6390, Next state: 0x80
2020-10-07 13:35:51 logStreamMetric(): Kinesis Video client and stream metrics:
        Overall storage byte size: 2097152
        Available storage byte size: 2055322
        Allocated storage byte size: 41830
        Total view allocation byte size: 691264
        Total streams frame rate (fps): 120
        Total streams transfer rate (bps): 16777216 (16384 Kbps)
        Current view duration (ms): 0
        Overall view duration (ms): 0
        Current view byte size: 0
        Overall view byte size: 0
        Current frame rate (fps): 120.000000
        Current transfer rate (bps): 16777216 (16384 Kbps)
2020-10-07 13:35:52 logStreamMetric(): Kinesis Video client and stream metrics:
        Overall storage byte size: 2097152
        Available storage byte size: 1851661
        Allocated storage byte size: 245491
        Total view allocation byte size: 691264
        Total streams frame rate (fps): 190
        Total streams transfer rate (bps): 16777216 (16384 Kbps)
        Current view duration (ms): 682
        Overall view duration (ms): 682
        Current view byte size: 200412
        Overall view byte size: 200412
        Current frame rate (fps): 190.551762
        Current transfer rate (bps): 16777216 (16384 Kbps)
2020-10-07 13:35:53 logStreamMetric(): Kinesis Video client and stream metrics:
        Overall storage byte size: 2097152
        Available storage byte size: 1488005
        Allocated storage byte size: 609147
        Total view allocation byte size: 691264
        Total streams frame rate (fps): 225
        Total streams transfer rate (bps): 16777216 (16384 Kbps)
        Current view duration (ms): 2242
        Overall view duration (ms): 2242
        Current view byte size: 563721
[... 96 lines cut ...]
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAWHMTBR7KOFSH7BW6/20201007/us-east-2/kinesisvideo/aws4_request, SignedHeaders=connection;cont
ent-type;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-sta
rt-timestamp;x-amzn-stream-name, Signature=cc9488245dfc720e73fed561796e95ca1ce99f0ff9a372e89af46e5a2c00662d
    connection: keep-alive
    content-type: application/json
    host: s-171c51c7.kinesisvideo.us-east-2.amazonaws.com
    transfer-encoding: chunked
    user-agent: AWS-SDK-KVS/2.0.1 GCC/6.3.0 Linux/4.9.37 armv7l
    X-Amz-Date: 20201007T133551Z
    x-amzn-fragment-acknowledgment-required: 1
    x-amzn-fragment-timecode-type: ABSOLUTE
    x-amzn-producer-start-timestamp: 1602077751.449
    x-amzn-stream-name: 6023A4BC43AC
2020-10-07 13:36:04 putStreamCurlHandler(): Network thread for Kinesis Video stream: 6023A4BC43AC with upload handle: 0 exited. http status: 40
8
2020-10-07 13:36:04 defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 2371213
2020-10-07 13:36:04 viewItemRemoved(): Reporting a dropped frame/fragment.
2020-10-07 13:36:04 defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 2371213. Dropped frame timecode in 10
0ns: 16020777531280000
2020-10-07 13:36:04 viewItemRemoved(): Reporting a dropped frame/fragment.
2020-10-07 13:36:04 defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 2371213. Dropped frame timecode in 10
0ns: 16020777531940000
2020-10-07 13:36:04 viewItemRemoved(): Reporting a dropped frame/fragment.
2020-10-07 13:36:04 defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 2371213. Dropped frame timecode in 10
0ns: 16020777532190000

MushMal, 2020-10-08:
This status code is not yet documented in the AWS docs:

It is defined in PIC: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L182

STATUS_MAX_FRAME_TIMESTAMP_DELTA_BETWEEN_TRACKS_EXCEEDED

This is an indicator that in a multi-track stream, the audio frames are being accumulated without a video frame. The AV sync algorithm for multi-track has to reorder some of the frames in order to produce a valid streamable fragment. The depth of the buffer is a few frames and this error happens when that buffer is being overrun. This happens in a real-life case primarily when you have the video media pipeline to stop or pause while the audio pipeline is still producing audio frames.

Try to investigate why your media pipeline stalls.

laoyouji007, 2020-10-09:
Thanks ! I will check the audio and video pipeline, Do you have any good debugging settings for me ?

**2020-10-07 13:35:59 commonHeapAlloc(): Allocating 15337 bytes failed due to heap limit
2020-10-07 13:35:59 putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp: 160207775946**

At the same time, there is an error prompt in the above log. What is the cause of this error prompt? Is there not enough memory? I can only allocate 2MB of space to kinesis for embedded devices. But in general, it can run normally, but after running for a period of time, there will always be this error.

MushMal, 2020-10-09:
Not sure what would be the best but you could enable verbose logging which would give you enough information about the frames, the track id and their timestamps. Check out the metrics for the stream: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/1fd494b1d1d557b18c826c62499fff09423987f5/src/KinesisVideoStream.cpp#L165

Regarding the heap alloc. This is an indication that you are getting your content store full. The latest commits in the master have a default behavior of pushing the tail frame out causing dropped frame rather than failing the putFrame call. Either way, you will be having pressures.

Here is the wip document describing the buffering: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/2825a0e8469a81fb44f853e7e2caa49993a52230/docs/buffering.md

You need to calculate the amount of buffer your application needs in terms of temporal and physical storage. You also might want to handle the pressures if the default policy is not good.

I am going to resolve this issue. Please don't hesitate to cut a new issue which targets a specific question or you need to get more help.

closed 2020-10-09 by MushMal