awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #587: [QUESTION] Timeout because operation too slow

FlorianRuen, 2020-10-12:
Hi there,

I'm using a Raspberry Pi to send video from RTSP camera to Kinesis Video Stream and I got an issue.

Sometimes, after 2 or 3 days running without any problem, the video feed not receive any video. So I try to get full logs from my KVS producer (on RPI), and I get this one, just before the issue:

`2020-10-05 05:50:53 [2667562080] WARN - curlCompleteSync(): curl perform failed for url https://s-96607c7a.kinesisvideo.eu-central-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
2020-10-05 05:50:54 [2724197472] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-96607c7a.kinesisvideo.eu-central-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIASGHSSNSCDTGHGD6M/20201005/eu-central-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowDEBUG:recording_hours:Call to is_in_working_hour with start time = : '' and end time ''
DEBUG:recording_hours:Call to is_in_working_hour with start time = : '' and end time ''
ledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=2161e13348b31cdaae27e40bac126668f52d087d7
2020-10-05 05:50:54 [2724197472] DEBUG - putStreamCurlHandler(): Network thread for Kinesis Video stream: 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy with upload handle: 745 exited. http status: 0
2020-10-05 05:50:54 [2724197472] WARN - putStreamCurlHandler(): Stream with streamHandle 28626152 has exited without triggering end-of-stream. Service call result: 599
2020-10-05 05:50:54 [2724197472] INFO - kinesisVideoStreamTerminated(): Stream 0x1b4cce8 terminated upload handle 745 with service call result 599.
2020-10-05 05:50:54 [2724197472] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
2020-10-05 05:50:54 [2724197472] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:50:57 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:50:59 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:50:59 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:02 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:03 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:04 [2667562080] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
2020-10-05 05:51:04 [2667562080] DEBUG - streamReadyHandler invoked
2020-10-05 05:51:04 [2667562080] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
2020-10-05 05:51:04 [2667562080] INFO - putStreamResultEvent(): Put stream result event. New upload handle 746
2020-10-05 05:51:04 [2667562080] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
2020-10-05 05:51:04 [2667562080] INFO - writeHeaderCallback(): RequestId: deb2d01b-76bc-e124-8946-4a73be5a62bf
2020-10-05 05:51:04 [2724197472] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 746
2020-10-05 05:51:04 [2724197472] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:06 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:10 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:11 [2667562080] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:18 [2667562080] WARN - curlCompleteSync(): curl perform failed for url https://s-96607c7a.kinesisvideo.eu-central-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
2020-10-05 05:51:34 [2724197472] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-96607c7a.kinesisvideo.eu-central-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIASGHSSNSCDTGHGD6M/20201005/eu-central-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=6ce2483ee21e865fcfe2b4c3ee828c6d8bc46ab63
2020-10-05 05:51:34 [2724197472] DEBUG - putStreamCurlHandler(): Network thread for Kinesis Video stream: 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy with upload handle: 746 exited. http status: 0
2020-10-05 05:51:34 [2724197472] WARN - putStreamCurlHandler(): Stream with streamHandle 28626152 has exited without triggering end-of-stream. Service call result: 599
2020-10-05 05:51:34 [2724197472] INFO - kinesisVideoStreamTerminated(): Stream 0x1b4cce8 terminated upload handle 746 with service call result 599.
2020-10-05 05:51:34 [2724197472] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
2020-10-05 05:51:34 [2724197472] DEBUG - Dropping frame with flag %u8256
2020-10-05 05:51:57 [2667562080] INFO - stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 0000000001b4cce8.
2020-10-05 05:52:37 [2667562080] DEBUG - streamClosedHandler invoked for upload handle: 18446744073709551615
2020-10-05 05:52:37 [2667562080] DEBUG - Reported streamClosed callback for stream handle 28626152. Upload handle 18446744073709551615
2020-10-05 05:52:37 [2667562080] INFO - Sending eos
2020-10-05 05:52:37 [2667562080] EOS reçu de l’élément « pipeline0 ».
Execution ended after 117:21:01.386293412
Définition du pipeline à PAUSED...
Définition du pipeline à READY (prêt)…
Définition du pipeline à NULL…
Libération du pipeline…
INFO - Freeing Kinesis Video Stream 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy
2020-10-05 05:52:37 [3069440016] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2020-10-05 05:52:37 [3069440016] DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty`

(I can translate french line to english if needed, but the problem came from the first one and the last one)

I don' really understand the source of the problem:

- Internet connexion list, so can"t send the stream ?
- Timeout after 3 days or key expire or something else ?

I try to get the video without KVS and save it a file, and there is no problems after 1 week and more. So the problem isn't my RTSP camera. 

This problem happen on production mode, so the RPI is wired before a firewall, maybe this is the source of my problem ?

If you wan give me some advices,

Waiting for an answer,

Kind regards,
Florian

MushMal, 2020-10-12:
The logs indicate a network timeout - the network client is set to timeout if there is no transfer in 30 seconds. This could happen either when the backend is unreachable somehow (timeout) or if there are no frames being produced.

From the small piece of log I could see that the state machine is attempting to recover and goes through the states and gets a new endpoint which makes me think that you don't have an issue with the networking but rather no frames are being produced. This said, I am not sure what your application does as I see printouts like this

2020-10-05 05:51:04 [2724197472] DEBUG - Dropping frame with flag %u8256

can you ensure your stream is constantly producing frames? If the frames stop being produced, the latest PIC code will not issue a new PutMedia request unless there is data in the buffer to send so it will keep waiting for PutFrame.

FlorianRuen, 2020-10-12:
As I saif il my first post, my producer is after a firewall on company network, so sometime there is a daily backup, and some networks streams are stopped to get the maximun bandwith for the backup.

This behavior can cause RTSP stream stop or reduce the bandwith from my producer to AWS.

There is a way to increase the 30 seconds to somethint bigger to avoid the producer stop when the backup is running ? (Maybe to set something like 5 hours because its the maximum time of the backup, 1 or 2 nights per week)

Thanks

MushMal, 2020-10-12:
The short answer is no as you could change the SDK limit but the backend has 30 second inactivity limit too.

Looking deeper, I would ask you to actually understand what the root cause is - is this a timeout caused by the SDK not able to submit the data to the backend or there is an issue with the application not producing frames into the SDK? It's easy to understand this if you enable debug logs or you look into the metrics.

Let's consider the first case - the SDK is unable to upload. In this case, your application should react on the network outage by perhaps slowing down, bit-thinning (dropping some fragments) or simply pausing until the network is back. There are many ways to handle it and your application should handle this. It would also work if you do nothing - it would simply overflow the buffer and start dropping frames. The state machine would keep trying to connect and send. It would pick up and start streaming when the network is back.

Let's consider the latter - the application is not producing frames into the SDK which causes the SDK to timeout. This scenario is called an "intermittent producer" scenario. In this case, (later commits) the SDK would stream out the existing buffer and after a timeout it would loop back into the ready state and pause there. Next time the application produces a frame (it will skip P-frames) the SDK will issue a PutMedia and start streaming.

disa6302, 2020-10-26:
@FlorianRuen , 

Assuming the question is answered, I am closing this issue. Feel free to open a new issue if you have any more questions.

closed 2020-10-26 by disa6302