awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #588: [QUESTION]How can I temporarily disable log output?

laoyouji007, 2020-10-13:
Hello !
When I run kinesis, a lot of log debugging information appears. Which method can temporarily disable log output?

logs like these:
2020-10-13 23:54:39 logStreamMetric(): Kinesis Video client and stream metrics: 
        Overall storage byte size: 3145728
        Available storage byte size: 3023388
        Allocated storage byte size: 122340
        Total view allocation byte size: 691264
        Total streams frame rate (fps): 2813
        Total streams transfer rate (bps): 623944 (609 Kbps)
        Current view duration (ms): 740
        Overall view duration (ms): 1950
        Current view byte size: 9533
        Overall view byte size: 71811
        Current frame rate (fps): 2813.798768
        Current transfer rate (bps): 623944 (609 Kbps)
2020-10-13 23:54:39 postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
2020-10-13 23:54:39 postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
2020-10-13 23:54:39 postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0

MushMal, 2020-10-13:
In C++ look for the settings for log4cplus. If you are using C producer and the default logger then look for SET_LOGGER_LOG_LEVEL macro

disa6302, 2020-10-19:
Closing this since the question is answered.

closed 2020-10-19 by disa6302