awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #590: [BUG]0x5200002e error happen when running kinesis-video-c-producer, and then the program will stop

laoyouji007, 2020-10-15:
When I started to run the kinesis-video-c-producer program, a memory allocation failure soon occurred, as shown in the log below. Due to the use of embedded SOC, the memory is relatively small, and the program memory allocation settings are as follows:
#define STORAGE_ALLOCATION_DEFRAGMENTATION_FACTOR                           1.2
#define DEFAULT_STREAM_COUNT                                                5
#define DEFAULT_DEVICE_STORAGE_SIZE                                         (3 * 1024 * 1024)

LOGS：
[20201015_14:33:45] 2020-10-15 06:33:44 logStreamMetric(): Kinesis Video client and stream metrics: 
[20201015_14:33:45]         Overall storage byte size: 3145728
[20201015_14:33:45]         Available storage byte size: 130170
[20201015_14:33:45]         Allocated storage byte size: 3015558
[20201015_14:33:45]         Total view allocation byte size: 691264
[20201015_14:33:45]         Total streams frame rate (fps): 233
[20201015_14:33:45]         Total streams transfer rate (bps): 12376000 (12085 Kbps)
[20201015_14:33:45]         Current view duration (ms): 11023
[20201015_14:33:45]         Overall view duration (ms): 11557
[20201015_14:33:45]         Current view byte size: 2834091
[20201015_14:33:45]         Overall view byte size: 2956118
[20201015_14:33:45]         Current frame rate (fps): 233.133822
[20201015_14:33:45]         Current transfer rate (bps): 12376000 (12085 Kbps)
[20201015_14:33:45] 2020-10-15 06:33:45 commonHeapAlloc(): **Allocating 13299 bytes failed due to heap limit**
[20201015_14:33:45] 2020-10-15 06:33:45 putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp: 16027436254639816 presentation timestamp: 16027436254639816
[20201015_14:33:45] 2020-10-15 06:33:45 commonHeapAlloc(): Allocating 13131 bytes failed due to heap limit
[20201015_14:33:45] 2020-10-15 06:33:45 putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp: 16027436254840323 presentation timestamp: 16027436254840323
[20201015_14:33:45] 2020-10-15 06:33:45 **Failed with status 0x5200002e**
[20201015_14:33:45] 
[20201015_14:33:45] 2020-10-15 06:33:45 freeKinesisVideoStream(): Freeing Kinesis Video stream.

MushMal, 2020-10-15:
The earlier commits had a default behavior of failing the putframe call on a heap limit error. The newer commits have a default behavior of pushing older frames out. Either way, if seems that your physical content store size is being overflown. This is an indication of a low bandwidth or no bandwidth at all.

Debug your application to see whether you have a bandwidth issue or if you need to change your encoder settings to allow for smaller frame size to fit into the buffer space you allocate.

This is not a bug so I am going to untag.

Please resolve this issue and cut a directed issues if you need more help

disa6302, 2020-10-19:
Closing based on tag. Feel free to reopen or create a new issue if you have questions.

closed 2020-10-19 by disa6302