awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #592: [QUESTION] Curl 32 bits asked when Curl 64bits is built

PierreSau, 2020-10-16:
Trying to compile KVS C++ producer sdk on windows 10 / VSC 2017 via the x64 native command tool as ml64.exe is needed.
I tried to compile via this command : cmake -G "NMake Makefiles" -DBUILD_GSTREAMER_PLUGIN=ON
After building project_libcurl, it was the turn to build libkvspic
And I got the following output. It's not an environmment error. Maybe I found a bug ? 
[image]
If anyone could help me I'll be grateful.
Thanks in advance.

lherman-cs, 2020-10-19:
@PierreSau it seems that the curl version that we're using is not compatible with your system, could you try to upgrade the curl version here, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/CMake/Dependencies/libcurl-CMakeLists.txt#L30?

PierreSau, 2020-10-26:
Hello Thanks for your answer and your help. I tried changing the line as you ask and it still doesn't work.

MushMal, 2020-10-26:
Are you using 32 or 64 bit command line flags? I believe when you install VS, it creates two shortcuts which have defines for 32 or 64 bit command line tools. You need to ensure you use the right command line.

PierreSau, 2020-10-26:
I am using x86 native tools Command Prompt.  If you have a better tool to build the project please let me know. But on windows 10 I was unsuccesful trying to use mingw64 via msys2

MushMal, 2020-10-26:
Use the 64bit version. x86 is setting env vars for 32 bit which confuses the libCurl cmake list. You can file an issue with that project to fix the dependency on the env vars

PierreSau, 2020-10-27:
Thank you I'll try that asap

MushMal, 2020-10-27:
Please reopen if you need help

closed 2020-10-27 by MushMal

keiwong0624, 2021-03-29:
Hi all, I am facing the same problem. I have tried both x64 native and x86 native VS command prompt but they do not work. Any idea to resolve this issue.?

Lepm11, 2021-09-07:
Hi @RW0624 did you solve the problem?