awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #593: raspberry pi auto start error how to solve it?

MAKER-park, 2020-10-18:
hello!
We combined the aws iot function with the corresponding code in one code to check the operation.
`import RPi.GPIO as GPIO
from time import sleep

servoPin          = 10
SERVO_MAX_DUTY    = 12
SERVO_MIN_DUTY    = 3

GPIO.setmode(GPIO.BOARD)
GPIO.setup(servoPin, GPIO.OUT)

servo = GPIO.PWM(servoPin, 50)
servo.start(1)

import subprocess
import os
from uuid import getnode as get_mac
import time, json, ssl
import paho.mqtt.client as mqtt
import random
from getmac import get_mac_address

ENDPOINT = '****'
THING_NAME = 'rpi_test'

read_serial_file = open('/home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build/serial.txt', mode='r')

serial = read_serial_file.read(4)

print(serial)

name='pettalk/'
sub = name+serial

print(sub)

count=0

#print(serial.read(4))
#print(type(serial.read(4)))
#serial = read_serial_file.read(4)
print(serial)

def on_connect(mqttc, obj, flags, rc):
        if rc == 0:
                print('connected!!')
                #name='pettalk/'
                #eth_mac = get_mac_address()
                #sub = name+serial
                print(sub)
                mqttc.subscribe(sub, qos=0)

def on_message(mqttc, obj, msg):
        if msg.topic == sub:
                payload = msg.payload.decode('utf-8')
                j = json.loads(payload)
                print(j['message'])
                degree=int(j['message'])
                duty=SERVO_MIN_DUTY+(degree*(SERVO_MAX_DUTY-SERVO_MIN_DUTY)/180.0)
                print("Degree: {} to {}(Duty)".format(degree, duty))
                servo.ChangeDutyCycle(duty)
                sleep(1)

##                if j['message'] == '180':
##                        print('yes!')
##                print(j['message'])

def make_movie():
 ps = subprocess.Popen(['iwconfig'], stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
 try:
    setting = 'cd Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build'
    output = subprocess.check_output(('grep', 'ESSID'), stdin=ps.stdout)
    print(output)
    #my_command = 'sudo AWS_ACCESS_KEY_ID=*****AWS_SECRET_ACCESS_KEY=****** AWS_REGION=ap-northeast-2 ./kvs_gstreamer_sample -w 1920 -h 1080 -f 30 -b 512000'
    my_command = 'sudo AWS_ACCESS_KEY_ID=AKIASWRVEBBPAE6V2ERP AWS_SECRET_ACCESS_KEY=***** ./kvs_gstreamer_sample '
    stream_name = 'pettalk_'
    #serial = read_serial_file.read(4)
    my_command_end = ' -w 1920 -h 1080 -f 30 -b 512000'
    #mac = str(get_mac())
    eth_mac = get_mac_address()

    print(eth_mac)
    os.system(setting)
    os.chdir("/home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build")
    os.system('ls')
    os.system(my_command+stream_name+serial+my_command_end)
    print(my_command+stream_name+serial+my_command_end)

 except subprocess.CalledProcessError:
    # grep did not match any lines
    print("No wireless networks connected")
    make_check()

def make_check():
 ps = subprocess.Popen(['iwconfig'], stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
 try:
    output = subprocess.check_output(('grep', 'ESSID'), stdin=ps.stdout)
    print(output)
    setting = 'cd Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build'
    #my_command = 'sudo AWS_ACCESS_KEY_ID=*** AWS_SECRET_ACCESS_KEY=***AWS_REGION=ap-northeast-2 ./kvs_gstreamer_sample -w 1920 -h 1080 -f 30 -b 512000'
    my_command = 'sudo AWS_ACCESS_KEY_ID=*** AWS_SECRET_ACCESS_KEY=****./kvs_gstreamer_sample '
    my_command_end = ' -w 1920 -h 1080 -f 30 -b 512000'
    stream_name = 'pettalk_'
    #mac = str(get_mac())
    eth_mac = get_mac_address()

    print(eth_mac)
    os.system(setting)
    os.chdir("/home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build")
    os.system('ls')
    #os.system(my_command+stream_name+serial)
    os.system(my_command+stream_name+serial+my_command_end)
    print(my_command+stream_name+serial+my_command_end)
    #print(my_command+stream_name+serial)

 except subprocess.CalledProcessError:
    # grep did not match any lines
    print("No wireless networks connected")
    make_movie()

mqtt_client = mqtt.Client(client_id=THING_NAME)
mqtt_client.on_connect = on_connect
mqtt_client.on_message = on_message
mqtt_client.tls_set('/home/pi/Desktop/awsiot/cer/CAG2.pem', certfile='/home/pi/Desktop/awsiot/cer/certificate.pem.crt',
        keyfile='/home/pi/Desktop/awsiot/cer/private.pem.key', tls_version=ssl.PROTOCOL_TLSv1_2, ciphers=None)
mqtt_client.connect(ENDPOINT, port=8883)
mqtt_client.loop_start() # threaded network loop

#make_movie()
while True:
		if count == 0:
			make_movie()
			count=1
	#make_movie()
	#time.sleep(2)
	#test = random.randint(0,180)
	#payload = json.dumps({'action': test})
	#mqtt_client.publish('test/1', payload, qos=1)

`

However, if i set the configuration so that the file can be automatically executed in /etc/profile, the following error occurs. How to fix it?

`
# /etc/profile: system-wide .profile file for the Bourne shell (sh(1))
# and Bourne compatible shells (bash(1), ksh(1), ash(1), ...).

if [ "`id -u`" -eq 0 ]; then
  PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
else
  PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games"
fi
export PATH

if [ "${PS1-}" ]; then
  if [ "${BASH-}" ] && [ "$BASH" != "/bin/sh" ]; then
    # The file bash.bashrc already sets the default PS1.
    # PS1='\h:\w\$ '
    if [ -f /etc/bash.bashrc ]; then
      . /etc/bash.bashrc
    fi
  else
    if [ "`id -u`" -eq 0 ]; then
      PS1='# '
    else
      PS1='$ '
    fi
  fi
fi

if [ -d /etc/profile.d ]; then
  for i in /etc/profile.d/*.sh; do
    if [ -r $i ]; then
      . $i
    fi
  done
  unset i
fi
sudo python3 /home/pi/Desktop/amazon-kinesis-video-streams-producer-sdk-cpp/build/petalk_mqtt_command.py
`

`socket.gaierror: [Errno -3] Temporary failure in name resolution`

[image]

MushMal, 2020-10-18:
I am not sure if this is intended for KVS forums or not. Could you perhaps debug your solution and see what the issue is and how is it related to KVS. Try to capture the debug logs and update the issue

MAKER-park, 2020-10-20:
> 이것이 KVS 포럼을위한 것인지 아닌지 잘 모르겠습니다. 솔루션을 디버깅하고 문제가 무엇이며 KVS와 어떤 관련이 있는지 확인할 수 있습니까? 디버그 로그를 캡처하고 문제를 업데이트하십시오.

> I am not sure if this is intended for KVS forums or not. Could you perhaps debug your solution and see what the issue is and how is it related to KVS. Try to capture the debug logs and update the issue

f you are using cpp, you simply use python to run the shell.

If you know how to see more of the error history, please tell me how to do that, and I'll look it up. As of now, the error details are all the error details shown in the photo above. 👍

MushMal, 2020-10-20:
I am sorry, I still don’t understand what the question is and how it is related to KVS. Please clarify.

MAKER-park, 2020-10-20:
> I am sorry, I still don’t understand what the question is and how it is related to KVS. Please clarify.
Thank you for quick response. But for now, it seems to be a problem between kvs and iot. Another function, servo motor, has been observed to run normally automatically.

It seems obvious that either kvs or aws iot is the problem. In my code, only the code for implementing the kvs aws iot function is added, and thus this problem occurs only in autorun. When you log in to ssh, the function is automatically executed normally. I am very confused as to which is the problem if the other is the problem.

MushMal, 2020-10-20:
Have you done any debugging yourself? Do you know how to split the problem into smaller ones so you can eliminate each one? Can you at least state the problem? How can we help you?

MushMal, 2020-11-08:
Closing due to inactivity

closed 2020-11-08 by MushMal