awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #594: [QUESTION] Clarification needed on AV Sync behavior

sharadrj, 2020-10-23:
Question from a customer - 
When pushing Audio/Video frames to the Producer SDK, does applications need to worry about A/V synchronization?
For example, what happens if we push a new key frame, but then push audio that corresponds with the previous GOP?
Does KVS handle this, or do we need to delay the new key frame until all the audio for the previous GOP has been pushed?

MushMal, 2020-10-23:
The question is not properly positioned. It really depends on the downstream playback components and how they handle the multitrack fragments. In general, the packaging - aka the fragmentation is a packaging concept and as such is a higher level concept. Depending on your downstream application, the multutrack package could be consumed by your application and it could make certain assumptions. In general, most of the playback systems will handle the multitrack packaging by simply unpacking it, running the playback AV sync of the player and submitting the encoded frames from the elementary stream to their respective decoder and renderer pipelines. 

This said, the packaging of the fragment has it's own limitation. 

* The fragment should be self-contained
* The frames should be ordered by their decoding timestamps - this comes from MKV specification
* KVS in addition has a limitation that each fragment should contain at least a single frame from the declared tracks (this is indeed a limitation currently)

In order to aid constructing a syntactically and semantically valid MKV fragment, the KVS SDK is able to reorder the audio video multi-track frames to produce valid fragments. PIC has a FrameOrderCoordinator that is able to accumulate and release frames with the key-frames. The internal buffer is defined to be 60 frames large. There is a check for the overflow:

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/src/FrameOrderCoordinator.c#L269

This could happen in cases when the audio encoder keeps producing audio frames while the video encoder encounters a large jitter - for example, at times software encoders could suffer from the CPU starvation.

Please let us know if you have any further information.

We are planning on including this information in the documentation

sharadrj, 2020-10-23:
Thanks for the reply, @MushMal.  Acting as the voice of the customer, I would like to rephrase the question - 

Applications pass the frames to KVS SDK with putKinesisVideoFrame(streamHandle, &frame) function. Does SDK imposes any constraints on the frames being passed to it for its internal processing (MKV packaging or anything else)? If yes, it will be great if we can list these constraints clearly.

MushMal, 2020-10-23:
The frames should be time-aligned (audio and vide source should use the same clock) in order to be interleaved into multi-track fragment. The frames can be produced independently by the media pipelines as long as the number of audio frames with a timestamp greater than the last video frame being produced does not exceed the default 60 frames. We use a queue for each track to accumulate the frames and produce in an interlaced fashion to ensure the increased order. (The logic also handles edge cases like EoFR, etc..).

Customers that have higher pressures on video pipeline could increase the default MAX number of frames to keep at the expense of larger heap memory allocation.

sharadrj, 2020-10-24:
Thanks this helps.  Few more questions - 

1. Are there any ordering constraints for timestamps for frames of  a track like DTS(n+1) > DTS(n)?
2. Are there any ordering constraints for timestamps  around fragment boundaries, like DTS_current_frame > DTS_last_key_frame
3. Are there any ordering constraints across tracks, like DTS_audio_frame >DTS_last_video_frame? 
4. Are there any constraints around DTS and PTS itself, like DTS(key frame) == PTS(key_frame)?
5. Does SDK uses DTS or PTS for reordering frames inside SDK?

MushMal, 2020-10-24:
1) In general DTS should be monotonically increasing and PTS could drift back as long as it's within the fragment (b-frames). Our backend can choke on the b-frames though on the reader APIs like HLS/MPEG-DASH. There are multiple edge cases that the code handles - for example, the granularity in MKV timecode scale could be identical but in SDK timecode scale of 100ns they could be different. The PutFrameCoordinator handles some of the edge cases we see in the real media pipelines, including some jittery ones like rtsp plugin for GStreamer. MKV specification states that the frames should be ordered in the decoding order. The MKV only has a single timestamp which corresponds to the PTS.
2) In the latest SDK we have relaxed the monotonicity rules. Though, our backend currently enforces strict monotonically increasing fragments which means that the PTS of the key frame of the fragment should be greater than the PTS of the last frame of the previous fragment. The SDK enforces greater or equal which is the right thing to do as many media pipelines treat frames as points in time rather than frames having durations.
3) The FrameOrderCoordinator handles the reordering up-to 60 frames in diff. 
4) For the type of audio encoders we use there should be no difference between the PTS and DTS. For video, there could be if we have b-frames. Key frames SHOULD have same dts and pts. However, in many media pipelines and scenarios we could be missing DTS or PTS for example, or, the PTS and DTS could be on a different clocks - the PTS could be the absolute clock whereas the DTS could be starting from 0. The GStreamer plugin handles some of the cases and the PIC handles some logic internally too - for example zero duration.
5) The PIC has different modes of operation. C Producer uses FRAME_ORDERING_MODE_MULTI_TRACK_AV_COMPARE_PTS_ONE_MS_COMPENSATE mode which is using PTS.

Unfortunately, due to the variability of the media pipelines and weird encoder implementations, we have to compensate in the GStreamer layer some of the behavior - for example, if the upstream is file-based then there are no DTS timestamps all together.

disa6302, 2020-11-23:
Closing this issue.

closed 2020-11-23 by disa6302