awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #595: Inconsistency in live video streaming

vickyskarthik, 2020-10-25:
I followed a AWS blog for controlling raspberrypi robot using AWS IOT and AWS Kinesis video stream - signaling channel to stream real time video from a Raspberrypi camera. The video is played in a website hosted through S3 bucket.  But I experience inconsistency in the video stream.  When the same webpage is browsed through different devices, the video is visible only in few systems while a black screen is most.  When I inspect the webpage it shows that the it reveals the packets from kinesis.

Kindly help me on how to solve this issue.
I have attached the images of both the screens one with live video and other with black screen.  

[image]
[image]

MushMal, 2020-10-25:
@vickyskarthik what steps have you taken to debug the issue? Debug logs? Any diff between working and non working? Any software changes? How can we help?

vickyskarthik, 2020-10-26:
I followed the steps and code provided in this blog
https://aws.amazon.com/blogs/compute/building-a-raspberry-pi-telepresence-robot-using-serverless-part-2/
And github
https://github.com/aws-samples/aws-serverless-telepresence-robot/tree/master/web-app

MushMal, 2020-10-26:
My question was about you doing debugging to understand what could be wrong. Your description (if I interpret it correctly) indicates that your solution works on some browsers and doesn’t work on others. What steps have you taken to debug it? Any debug logs to show you what might be wrong? Can you open the webrtc internals in the browser?

vickyskarthik, 2020-10-26:
I have attached the webrtc internals logs for both the cases-when the video is visible and when it is not.  The logs are from different devices running the same OS and chrome browser of version 86.0.4240.111.  
Even in chrome, the video is not always visible and requires a restart.  In some devices, the video is not visible even after a restart in chrome
[webrtc_internals_dump with video.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5436741/webrtc_internals_dump.with.video.txt)
[without_video.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5436742/without_video.txt)

.

disa6302, 2020-10-26:
This is webrtc related. Do you mind closing the issue here and opening it in the appropriate repo?

vickyskarthik, 2020-10-26:
Ok. But the video available in the kinesis console -media playback viewer is also inconsistent. Even when I check the video in two systems simultaneously in console, it is visible in one and not in the other for the same user.

disa6302, 2020-10-26:
It should be. 
Please verify the details you add in the test page are correct. The stream name, client Id if any and the credentials, etc.