awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #596: [QUESTION] Operation too slow (reopen)

FlorianRuen, 2020-10-27:
Hello There,

I'm openning a new issue because the other one was closed (due to inactivity on my side). But there is some news on the case.
The old case was : [https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/587#event-3921368932](this one)

As @MushMal asked on the previous case, I've a video capture of the logs on what happend. As the logs said, it's a backend timeout due to operation too slow. There is some "dropping frame", but my RTSP source is reacheable if I use a video player like VLC or something else.

Here is the video screen : [uploaded on S3](https://eyesr-files-partage.s3.eu-central-1.amazonaws.com/2020-10-27+07-48-59.mkv

Do you have any further informations ?

Regards,
Florian
)

MushMal, 2020-10-27:
The logs I see seem to indicate that you have a network connection issue. 

* What asset you are using and which version?
* Have you done any modification on top of the stock samples?
* Perhaps enable VERBOSE log level to get more spew.
* Is your application using fragment metadata or only frames?
* What's the produced stream? H264? What's the media pipeline?

Could you also use the latest commit of the SDK? We have added some fixes for the edge cases around intermittent producer and metadata.

FlorianRuen, 2020-10-29:
@MushMal 

I will try to pull the latest commit of the SDK, I need to built it again (it runs on the Raspberry Pi)
For the VERBOSE, I just need to get the GST_DEBUG:=VERBOSE or anything else ?

The produced stream is H264 and the media pipeline is the following:

```
gst-launch-1.0 rtspsrc location=[RTSP_URL] short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name=[NAME]          storage-size=256 access-key=[KEY] secret-key=[SECRET] aws-region=eu-central-1
```

My application just send the frame over the pipeline (I've some checks or interactions to check if we need to send the stream or not, or if we need to stop the pipeline depending on hours, but no more)

MushMal, 2020-10-29:
GST_DEBUG will control the GStreamer verbosity. You will need to control the verbosity of the SDK. It seems that you have GStreamer which uses C++ producer which uses log4cplus which you can configure following this: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/b777e86bef634922c8062e051bd39e89468fa547/docs/macos.md#how-to-configure-logging-for-producer-sdk-sample-applications

You need to run your own experimentation and look through your logs to understand your pressures on your system. The gstreamer plugin is using the default continuous retry policy that attempts to reset the connection on the latency pressures but simply will drop the frames if your buffer won't recover on network congestion.

FlorianRuen, 2020-11-01:
@MushMal I will check the complete logs on tomorrow morning, I will get back on this ticket with more informations very soon

MushMal, 2020-11-08:
Closing this issue for now. Please cut a new issue if it proves to be related to KVS and not an issue with the staleness in the media pipeline

closed 2020-11-08 by MushMal