awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #598: [QUESTION] How to use Library with AWS Lambda under JAVA

Julchenwsf, 2020-10-30:
Include kinesis-video-stream-producer library in JAVA AWS LAmbda project

Hello,

I build these library under an EC2 instance from "amzn-ami-hvm-2018.03.0.20181129-x86_64-gp2" which is the environment from AWS lambda. After this I try to add the resulted .so file to my aws lambda java project by including it into the Library path. But when i upload it to Lambda it says "unsatisfied Link error". Do you have an Idea why this is not working?

Best regards Julia

hassanctech, 2020-10-30:
Can you run `ldd` on the so file to see what libraries it needs to link to?  Here is an example we have using the C library in Java with JNI:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java

The readme in the "Prerequisites" section explains if the pre-build libs don't work, how you can build them yourself.

In particular you'll find the .so files here for various platforms: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/tree/master/src/main/resources/lib

MushMal, 2020-11-08:
I am going to close this issue as there are no action items. Please cut a new issue with more scoped questions or if you have an issue then please provide detailed information and the verbose logs.

closed 2020-11-08 by MushMal