awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #601: Add clock display to the webcam stream

arnamoy10, 2020-11-02:
Hello;

I am planning to display clock on top of my webcam feed using `clockoverlay` option from gstreamer.

I see the custom pipeline creation in the `gstreamer_live_source_init ()` function, but I am not sure where and how to add the option to the pipeline for clock display.

Can anybody help me do this?

Thank you

hassanctech, 2020-11-02:
Yes so the `clockoverlay` element works on uncompressed video, so you'll need to add this element *after* your src element but *before* the encoder element.  If you have already encoded video, you can't use the `clockoverlay` element without decoding the video first.

As an example, for this line:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/76e697c13e61de4fda29e0ec050339532fdfb8ea/samples/kvs_gstreamer_sample.cpp#L685

`gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=I420,width=1280,height=720,framerate=15/1 ! x264enc pass=quant bframes=0 ! video/x-h264,profile=baseline,format=I420,width=1280,height=720,framerate=15/1 ! matroskamux ! filesink location=test.mkv`

Would become:

`gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=I420,width=1280,height=720,framerate=15/1 ! clockoverlay ! x264enc pass=quant bframes=0 ! video/x-h264,profile=baseline,format=I420,width=1280,height=720,framerate=15/1 ! matroskamux ! filesink location=test.mkv`

Closing, feel free to re-open if you still need help

closed 2020-11-02 by hassanctech

arnamoy10, 2020-11-02:
Thanks for the answer.  Could you please point out in the source code file where the `x264enc pass=quant bframes=0` part is added to the pipeline?  It will be of great help.  Sorry I am new to gstreamer

hassanctech, 2020-11-02:
That comment is an example, in the code we do something a bit different, only the bframes=0 part is set:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/76e697c13e61de4fda29e0ec050339532fdfb8ea/samples/kvs_gstreamer_sample.cpp#L811

There's some logic to determine which encoder to use, here is where `x264enc` gets selected:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/76e697c13e61de4fda29e0ec050339532fdfb8ea/samples/kvs_gstreamer_sample.cpp#L704

This will of course all need to be tuned based on your media pipeline, but the basic gist of it is once you get the media frames (uncompressed) you put the `clockoverlay` element before the encoding.  If your media source is producing already encoded frames and you don't have access to the raw frames, then you have to check the source element to see if it has properties that allow it to do the clock overlay, alternatively sometimes the webcam UI will provide an option to to a clock overlay on the frame.