awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #602: CMake can't find gstreamer installed with homebrew on Mac Catalina

virenbajaj, 2020-11-02:
## Problem

I'm trying to build the sdk by following the README on my mac OS Catalina 10.15.7, 
but the command: 
`cmake .. -DBUILD_DEPENDENCIES=OFF -DBUILD_GSTREAMER_PLUGIN=ON`
fails with error: 
```
-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
```

Full output:
```
cmake .. -DBUILD_DEPENDENCIES=OFF -DBUILD_GSTREAMER_PLUGIN=ON
-- Kinesis Video Cpp Producer path is <MY_PATH>/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is <MY_PATH>/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Configuring done
-- Generating done
-- Build files have been written to: <MY_PATH>/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No configure step for 'libkvscproducer-download'
[ 33%] No build step for 'libkvscproducer-download'
[ 44%] No install step for 'libkvscproducer-download'
[ 55%] No test step for 'libkvscproducer-download'
[ 66%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is <MY_PATH>/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Configuring done
-- Generating done
-- Build files have been written to: <MY_PATH>/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No configure step for 'libkvspic-download'
[ 33%] No build step for 'libkvspic-download'
[ 44%] No install step for 'libkvspic-download'
[ 55%] No test step for 'libkvspic-download'
[ 66%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
-- Bitness 64 bits
found log4cplus.
-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
CMake Error at /opt/local/share/cmake-3.12/Modules/FindPkgConfig.cmake:436 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.12/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
  CMakeLists.txt:190 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "<MY_PATH>/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
```

## Tries:

### 1 
I added the following code to CMakeLists.txt right after line `188 if(BUILD_GSTREAMER_PLUGIN)`
```
set(PKG_CONFIG_USE_CMAKE_PREFIX_PATH 1)
set(CMAKE_PREFIX_PATH /usr/local/Cellar/)
```
### 2
Followed [this issue](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/369) and did 
```
export PKG_CONFIG_PATH=/usr/local/Cellar/gstreamer/1.16.2/lib/pkgconfig/:/usr/local/Cellar/gst-plugins-base/1.16.2/lib/pkgconfig/
```
before running cmake. It didn't work.  I also reinstalled gstreamer as the fix by @disa6302 suggested. It didn't work.
### 3
Before this, I was getting errors while installing **log4cplus** and **gtest**, despite running the commands given in the README.
I rectified the gtest error by following [this](https://stackoverflow.com/a/42707478) post on StackOverflow.
The solution was basically downloading and building gtest from the github repo and then manually moving the generated lib and include files to /usr/local/lib and /usr/local/include.
So by following the same pattern I moved the lib and include files from the Homebrew Cellar: /usr/local/Cellar to /usr/local as so: 
`cp -r /usr/local/Cellar/gstreamer/1.18.1/lib/ /usr/local/lib` and
`cp -r /usr/local/Cellar/gstreamer/1.18.1/include/ /usr/local/include`

I was able to cmake without errors, then make successfully **BUT** 
running 
`gst-inspect-1.0 kvssink`
gave the error:
`(process:74572): glib-gobject-critical **: 15:45:32.920: g_param_spec_pool_lookup: assertion 'pool != null' failed`

**Please help me understand how to make CMakeLists look for gstreamer in the Homebrew Cellar.**

MushMal, 2020-11-02:
None of these should be required. We are not officially supporting Catalina as some of the 3rd party dependencies (such as GStreamer) are having issues with the file system protection. Disable the file system protection and retry installing.

Follow the readme to set the env variables for LD_LIBPATH and the plugin path for the GStreamer.

Please resolve once you've done.

virenbajaj, 2020-11-03:
I disabled file system protection:
```
csrutil status
System Integrity Protection status: disabled.
```
uninstalled gstreamer:
`brew uninstall --ignore-dependencies gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly gst-libav`

then reinstalled it. 

But I still get the same error.

MushMal, 2020-11-03:
Check out other Catalina related issues here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/search?q=catalina&type=issues

For more info please reach out to GStreamer forums to get help installing and configuring.

One thing I noticed is that you have not installed GStreamer-app. Check out the Travis ci script in the root directory to properly install the prerequisites and to configure them

MushMal, 2020-11-08:
Closing due to staleness. Please try to resolve the issue with the GStreamer using GStreamer forums. Please cut a new issue with KVS specific questions.

closed 2020-11-08 by MushMal