awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #605: [QUESTION] H264+AAC sync see error code status: 0x30000005

crcsylvia, 2020-11-05:
Hello there,

I want to integrate audio track with codec AAC and sometimes I see error code 0x30000005(STATUS_CONTENT_VIEW_INVALID_TIMESTAMP). I assume it's the reason why the video and audio on AWS web console play intermittently (audio and video can not play smoothly).
Other information: SDK version v2.0.1 & with h264 video only stream (media from surveillance camera embedded system) works fine.

Here are the questions that I need your kindly help:
1. (Ref: [Related SDK code: ContentView.c Line 460](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/75087f5a90a02a47191c9278cfec329e09535e98/kinesis-video-pic/src/view/src/ContentView.c))
`CHK(pHead->timestamp + pHead->duration <= timestamp, STATUS_CONTENT_VIEW_INVALID_TIMESTAMP); `
Where does pHead->timestamp & pHead->duration & timestamp from?
 Looks like its different from the pts that I provide in each frame.
2. #594 
> In the latest SDK we have relaxed the monotonicity rules. Though, our backend currently enforces strict monotonically increasing fragments which means that the PTS of the key frame of the fragment should be greater than the PTS of the last frame of the previous fragment. The SDK enforces greater or equal which is the right thing to do as many media pipelines treat frames as points in time rather than frames having durations.

Does audio frames check timestamp of video track in fragments?
Is audio or video timestamp monotonically in their own track enough?

3. Do the SDK and AWS web console support the case which video or audio unexpectedly drop frames? 

4. Do you have another hint that can help to check why the 0x30000005 happened?

MushMal, 2020-11-05:
@crcsylvia older version of the SDK had a more stringent requirements for the monotonicity and continuity as you pointed out. The main idea is to ensure that we can produce syntactically valid fragments. The check in the latest version is to simply insure that the fragments don't overlap. Previously, we would take into account the frame timestamp + it's duration to not overlap with the next frame.

In your streaming case, are you using audio only stream? If so, you need to ensure that the frame timestamps are monotonically increasing and they have timestamps that don't get "rounded" down to the same value as the final stage of packaging requires the timecode scale conversion which by default is 1ms. What this means is that if you have two frames with less than a ms timestamps, they might end up having the same timestamps. You can also specify 0 for the duration of the frames.

If you use multi-track, you most likely use the frame order coordinator that handles some of the massaging, de-duping and ordering of the audio/video frames to achieve streamable valid fragment.

Your build is a little outdated so we would recommend upgrading to the latest release, run your scenario with the information I just provided and get more data out.

To answer your questions

1) The timestamps are adjusted to the timecode scale. In case of the frame order coordinator, the value of the timestamp can be adjusted up to ensure no overlap. Also, the frames can be reordered between the tracks.
2) There are complex rules governing on how we do av/sync. The main assumption is that the tracks will produce frames on the same timelines so they will be possible to av/sync. GStreamer plugin handles some timestamp changes in order to bring them to the same timeline.
3) Not sure what you mean by suddenly drop frames. If you mean by the producer or the network related issues result in no fragments being ingested (as the console playback is based on fragmented playback) then yes, it can skip the missing fragments.
4) That's the only one. The reason you are hitting this error is that your pipeline does not produce monotonically increasing PTS for the frames. 

Please resolve if you have no further questions

crcsylvia, 2020-11-06:
@MushMal

crcsylvia, 2020-11-06:
@MushMal Thanks for your response. There are some questions that still need to be clarified and we will try to build with the latest version of SDK to get more information. 
>The check in the latest version is to simply insure that the fragments don't overlap.

In our case(multi-track stream), we can only make sure video or audio monotonic independently. Ex: an audio frame might be input after the previous fragment is ended (audio track timestamps are still monotonic). Will this audio frame be dropped in the latest version of SDK? Do we still need to implement our own order coordinator after we update to the latest version? Or the order coordinator inside SDK would handle this case for us?
> 3. Not sure what you mean by suddenly drop frames.

There might be frame drop in one fragment.

MushMal, 2020-11-06:
Frame order coordinator will take care of ordering as long as the frames are on the same timeline - aka use the same clock. They don't need to be monotonically increasing with respect to each track as long as they are monotonically increasing within the track. The frame order coordinator has a buffer of 60 frames - this is the max "drift" that it can accommodate when blocking audio track in order to get the frame from video. You can check FrameOrderCoordinator for more info on the implementation of the AV/Sync.

Regarding the "frame drop". I am still not sure I understand the question. One of the current limitations our ingestion backend imposes is that each fragment should have at least one frame from each track.

MushMal, 2020-11-08:
Any update? Could you please close this issue if you have no further questions related to this issue?

crcsylvia, 2020-11-09:
We will try to build with the latest version of SDK to get more information, thank you!

closed 2020-11-09 by crcsylvia

wlinInspire, 2020-11-15:
@crcsylvia Did you solve the issue with the latest version? I have the same issue that uploading H264 +AAC gives error code 0x30000005 but H264 only does not give this error.

crcsylvia, 2020-11-25:
@MushMal 
After I update to sdk v3.0.0, I receive error FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS which I didn't meet in v2.0.1. I have make sure that each frame's pts and dts are monotonically. Need your help to provide other possibilities. Thank you very much.
I collect video dumps & partial log as attachment (full log prefer not to go public, but can be sent by mail). 
Note:
1. I also test the scenario with audio (also not smoothly play), but I guess it might due to the same cause.
2. log settings: LOGLEVEL=TRACE KVS_DEBUG_DUMP_DATA_FILE_DIR=/home/recording 

[log.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5594587/log.zip)

MushMal, 2020-11-26:
@crcsylvia there are changes between v2.0.1 and v3.0.0 and the requirements are actually less stringent. It is has more laxed checks on the client side but the backend service strictly enforces monotonicity of the stream. What this means is that the timestamp of the first frame of the fragment should be GREATER than the timestamp of the last frame of the previous fragment. The SDK enforces GREATER or EQUAL.

Can you check your pipeline and see what drives the timestamps? Generally speaking, every frame your pipeline produces should have incremental timestamps which would ensure monotonically increasing timestamps in the fragments too.

wlinInspire, 2020-11-30:
@MushMal 

I have the same issue that streaming video (H264) + audio (AAC) gives warning "putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client." with error code 0x30000005 but video-only does not give any warning. Below is the log comparison. A couple of observations

* The framerate is 100 for audio-video while it is 25 for video-only.
* The frame ordering mode is 3 for audio-video while it is 0 for video-only.

[image]

The extracted log for both tracks as well as the waring occurrence is attached below. Tried to debug this for a long time but have no clue yet. Please let me know if you know what was happening. Thanks.
```
[2020-11-30 20:48:26:564.945 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693065648992, dts: 0, duration: 0, size: 23, trackId: 2, isKey 0
[2020-11-30 20:48:26:565.102 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693065881192, dts: 232199, duration: 0, size: 6, trackId: 2, isKey 0
[2020-11-30 20:48:26:565.176 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693066113391, dts: 464399, duration: 0, size: 6, trackId: 2, isKey 0
[2020-11-30 20:48:26:565.228 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693066345591, dts: 696598, duration: 0, size: 6, trackId: 2, isKey 0
[2020-11-30 20:48:26:565.279 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693066577791, dts: 928798, duration: 0, size: 6, trackId: 2, isKey 0
[2020-11-30 20:48:26:565.330 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693066809990, dts: 1160997, duration: 0, size: 6, trackId: 2, isKey 0
[2020-11-30 20:48:26:565.378 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693067042190, dts: 1393197, duration: 0, size: 6, trackId: 2, isKey 0
[2020-11-30 20:48:26:565.427 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693067274389, dts: 1625396, duration: 0, size: 6, trackId: 2, isKey 0
[2020-11-30 20:48:26:586.919 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693067506589, dts: 1857596, duration: 0, size: 122, trackId: 2, isKey 0
[2020-11-30 20:48:26:603.160 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693067738788, dts: 2089795, duration: 0, size: 237, trackId: 2, isKey 0
[2020-11-30 20:48:26:623.280 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693067970988, dts: 2321995, duration: 0, size: 242, trackId: 2, isKey 0
[2020-11-30 20:48:26:660.235 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693067888788, dts: 2334714, duration: 0, size: 109874, trackId: 1, isKey 0
[2020-11-30 20:48:26:671.253 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693068018788, dts: 2365319, duration: 0, size: 11284, trackId: 1, isKey 0
[2020-11-30 20:48:26:671.345 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693068203187, dts: 2554195, duration: 0, size: 504, trackId: 2, isKey 0
[2020-11-30 20:48:26:671.494 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693068348782, dts: 2688611, duration: 0, size: 8996, trackId: 1, isKey 0
[2020-11-30 20:48:26:671.572 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693068435387, dts: 2786394, duration: 0, size: 13, trackId: 2, isKey 0
[2020-11-30 20:48:26:696.951 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693068667586, dts: 3018594, duration: 0, size: 151, trackId: 2, isKey 0
[2020-11-30 20:48:26:724.203 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693068688753, dts: 3025358, duration: 0, size: 9676, trackId: 1, isKey 0
[2020-11-30 20:48:26:724.611 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693068899786, dts: 3250793, duration: 0, size: 276, trackId: 2, isKey 0
[2020-11-30 20:48:26:743.495 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693069018709, dts: 3360747, duration: 0, size: 10673, trackId: 1, isKey 0
[2020-11-30 20:48:26:743.700 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693069131986, dts: 3482993, duration: 0, size: 257, trackId: 2, isKey 0
[2020-11-30 20:48:26:763.137 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693069348621, dts: 3692316, duration: 0, size: 12509, trackId: 1, isKey 0
[2020-11-30 20:48:26:763.299 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693069364185, dts: 3715192, duration: 0, size: 270, trackId: 2, isKey 0
[2020-11-30 20:48:26:796.533 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693069596385, dts: 3947392, duration: 0, size: 295, trackId: 2, isKey 0
[2020-11-30 20:48:26:816.190 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693069688477, dts: 4035650, duration: 0, size: 24296, trackId: 1, isKey 0
[2020-11-30 20:48:26:816.399 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693069828584, dts: 4179591, duration: 0, size: 239, trackId: 2, isKey 0
[2020-11-30 20:48:26:834.282 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693070018300, dts: 4362831, duration: 0, size: 23057, trackId: 1, isKey 0
[... cut ...]
[2020-11-30 20:48:33:537.037 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693137630852, dts: 71981859, duration: 0, size: 262, trackId: 2, isKey 0
[2020-11-30 20:48:33:552.066 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693137863051, dts: 72214058, duration: 0, size: 479, trackId: 2, isKey 0
[2020-11-30 20:48:33:556.113 GMT] [AWS Kinesis] [TRACE] putKinesisVideoFrame(): debug frame info pts: 16067693138095251, dts: 72446258, duration: 0, size: 8, trackId: 2, isKey 0

```

MushMal, 2020-11-30:
Please ensure the correct frame order coordinator mode is used. Set the durations of the frames to 0. Ensure the video and audio timestamps are on the same timeline. The audio/video sample is a good starting place. Dump out the frame timestamps and go over them to ensure there are no back-drifts in timestamps for a given track.

Unfortunately, I won't be able to look through your frame timestamps myself.

Please cut a new issue (as this is closed) if you are able to reproduce any issue that's related to KVS itself and not to your media pipeline.