awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #606: 'MAKE_EXE-NOTFOUND' is not recognized as an internal or external command, operable program or batch file.[QUESTION]

daling11, 2020-11-07:
When I using cmake to build the C++ producer sdk on windows, I have this error message:

[image]

This is my log file:

[CMakeOutput.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5504347/CMakeOutput.log)

What is 'MAKE_EXE-NOTFOUND' command? I installed the make.exe but it didn't work. Does anyone has this problem before?

MushMal, 2020-11-07:
I have not seen any issues like this. Could you make sure you have a clean build with the correct dependencies set. We have a windows Travis.ci build and the script that you can check out. Also, try to shorten the path and retry

MushMal, 2020-11-08:
Any updates on this issue?

daling11, 2020-11-08:
Sorry I forgot to update. I try to use X64 Native Tools Command Prompt instead of Cmake Gui and the problem fixed. Thank you very much for your help!

closed 2020-11-08 by daling11

aqiao, 2023-08-20:
@daling11 , the ` X64 Native Tools Command Prompt` you mentioned is the Windows build-in command line prompt?, i faced the same issue, and i used the Windows default command line prompt with administrator privilige