awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #608: [QUESTION] unable to stream more than 15 rtsps

hadi-alnehlawi, 2020-11-08:
Hi,

I am using the linux build producer and running the sample: 
`./kvs_gstreamer_multistream_sample`

it works fine for 15 rtsp camers, adding the 16th one it would get the below error:

`./kvs_gstreamer_multistream_sample camera_rtsp cameras.txt`

```
[ERROR] [08-11-2020 09:55:18:146.779 GMT] createKinesisVideoStream(): operation returned status code: 0x52000001
[ERROR] [08-11-2020 09:55:18:146.800 GMT] createKinesisVideoStreamSync(): operation returned status code: 0x52000001
[ERROR] [08-11-2020 09:55:18:146.872 GMT]  Unable to create Kinesis Video stream. new_16 Error status: 0x52000001
terminate called after throwing an instance of 'std::runtime_error'
  what():   Unable to create Kinesis Video stream. camera_rtsp_16 Error status: 0x52000001
Aborted (core dumped)
```

is there any limitation on the number of rtsp camers to be streamd?

MushMal, 2020-11-09:
The error you are getting is defined here:

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L68

Modify this line:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/DefaultDeviceInfoProvider.cpp#L23

Or, better yet, insert the following in LN 131 here:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_multistream_sample.cpp#L131

device_info.streamCount = YOUR_APPLICATION_REQUIRED_MAX_STREAM_COUNT;

closed 2020-11-09 by MushMal

bhavikapanara, 2020-12-31:
Hi @hadi-alnehlawi 

I am also looking for streaming from multiple RTSP sources. Can you please help me to know how you have streamed 15 RTSP sources into KInesis Video Stream?

Thanks a lot,
Bhavika

hadi-alnehlawi, 2021-01-01:
Hi @bhavikapanara,

Yes, you need to change the `#define DEFAULT_MAX_STREAM_COUNT        16` in the location:
`../amazon-kinesis-video-streams-producer-sdk-cpp/src/DefaultDeviceInfoProvider.cpp`

The default value is 16, and then build repo again.

Thanks,