awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #609: Streaming to KVS using gst and iot-certificate [QUESTION]

indoorrobot, 2020-11-09:
Hi,
I am trying to stream to KVS from an IoT device without the need of settings AWS credentials.

I followed https://docs.aws.amazon.com/kinesisvideostreams/ and now using the following command:

gst-launch-1.0 rtspsrc location="rtsp://127.0.0.1:$FRONT_PORT/front" short-header=TRUE user-id="$FRONT_USERNAME" user-pw="$FRONT_PASS" ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink name=sink stream-name="_fc" iot-certificate="iot-certificate, endpoint=XXXXXXXXX.credentials.iot.eu-central-1.amazonaws.com, cert-path=$PATH_TO_CERT/certificate.pem.crt,key-path=$PATH_TO_CERT/private.pem.key,ca-path=$PATH_TO_CERT/rootCA.pem, role-aliases=KvsCameraIoTRoleAlias" retention-period=4320

What I get in response is:

Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
DEBUG - Creating IoT auth callbacks.
2020-11-02 23:45:00 [139930241742592] 2020-11-02 21:45:00 ERROR   blockingCurlCall(): Curl call response failed with http status 403
ERROR - Unable to create Iot Credential provider. Error status: 0x15000011
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:sink: Could not initialize supporting library.
Additional debug info:
/amazon-kinesis-video-streams-producer-sdk-cpp/src/gstreamer/gstkvssink.cpp(1375): gst_kvs_sink_change_state (): /GstPipeline:pipeline0/GstKvsSink:sink:
Failed to init kvs producer. Error: Unable to create Iot Credential provider. Error status: 0x15000011
2020-11-02 23:45:00 [139930241742592] Setting pipeline to NULL ...
Freeing pipeline ...

the policy json I am using for KvsCameraIoTRoleAlias:

{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": "kinesisvideo:*",
            "Resource": "*"
        }
    ]
}

What am I missing?

MushMal, 2020-11-09:
You are doing the right thing it seems but you are getting permissions denied likely due to the policy document. Here is the actual documentation to generate the IoT credentials with the step-by-step instructions and samples.

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html

indoorrobot, 2020-11-09:
Hi @MushMal thanks for the quick reply! :)

This is what I followed and get the above response... I do not use the Resource Tag in the policy json (since the stream name is customized), is this required?

MushMal, 2020-11-09:
That's a good question... Not entirely sure as I found out that the resulting policy is very "sensitive" to the JSON format. Do you mind following this guide very closely to get this working and then modify to your needs.

disa6302, 2020-11-23:
@indoorrobot , 

Any updates?

disa6302, 2020-12-01:
Closing since it has gone stale.

closed 2020-12-01 by disa6302

clogwog, 2021-11-04:
i've got the same issue https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/781
seems to be related to not haing the stream-name being the same as the thing name

clogwog, 2023-08-26:
`gst-launch-1.0 -e filesrc location="/path/to/mp4/file/with/h264encoding.mp4" ! qtdemux name="mux1" mux1.video_0 ! h264parse config-interval=-1 ! "video/x-h264,stream-format=avc,alignment=au" ! kvs1.video_0 kvssink streaming-type=offline file-start-time=[epochtimeofwheretostorethisbitofvideointhestream] name=kvs1 storage-size=512 iot-certificate="iot-certificate,endpoint=xx.credentials.iot.us-west-2.amazonaws.com,cert-path=xxx-certificate.pem.crt,key-path=xxx-private.pem.key,ca-path=AmazonRootCA1.pem,role-aliases=YourRoleAlias,iot-thing-name=YourThingName" stream-name="YourStreamName" `

is working for me (i have a different way of authorisation though as i'm using greengrass to get a session kye, but the iot-certificate was working for us before as well)

pretty sure it doesn't make any difference, but i still have an idea in my head that the stream-name needs to be the last parameter to the kvssink (it makes no sense, but i thought i'd mention it)

clogwog, 2023-08-26:
> what is it the
> <awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/609/1694143207
> @github.com>

It’s the epoch time of where you want to upload that particular file to in
the stream. (This example is for an upload of a file in offline mode)

Checkout gst-describe kvssink
For a description of all parameters.