awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #611: Issues with using gstreamer to push RTSP stream to KVS

DManthoko, 2020-11-10:
## Summary of Issue

I would like to use the gstreamer plugin for KVS to push an RTSP stream to KVS. I installed the C++ producer SDK and the relevant libraries required for gstreamer but ran into an issue when running the gstreamer command. 

For testing, I used the RTSP URL provided on Wowza - rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov. Using VLC media player, I was able to play the stream. I followed the guide on the KVS documentation and on the ReadMe for this repository to install all the required libraries. However, when I run the gstreamer command, the RTSP stream does not get pushed to KVS. 

## Steps to reproduce issue

All my testing was done on an Ubuntu 18.04.5 LTS based EC2 instance. 

Before installing everything, ran `sudo apt-get update && sudo apt-get upgrade`

Installed the prerequisites as mentioned in [1]:

- `sudo apt-get install git cmake libtool libtool-bin  automake bison g++ curl pkg-config flex openjdk-8-jdk`
- Set JAVA_HOME environment variable to /usr/lib/jvm/java-8-openjdk-amd64
- Also downloaded the SFSRootCAG2.pem and saved that in the /etc/ssl folder
-  In [1], it is mentioned that you must run the build script `install-script`, however, when I downloaded the SDK I did not find such a file. Even after running cmake and make, I did not see this file. 

Ran the command shown in the ReadMe for this GitHub that specifiues what you need to install in order to use gstreamer.

`sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
`

After installing the above, I proceeded to build the SDK

- mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE

- After running the above command, ran `make` in the build directory
- Exported the path variables GST_PLUGIN_PATH and LD_LIBRARY_PATH by specifying them in my .bashrc file. 
- When I ran gst-inspect-1.0 kvssink, I got some information about the plugin

gst-launch-1.0 rtspsrc location="rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov" short-header=TRUE ! kvssink stream-name="<KVS-STREAM-NAME>" storage-size=512 access-key="<ACCESS-KEY>" secret-key="<SECRET-ACCESS-KEY>" aws-region="eu-west-1"

After the running the above command, this is what I see 

```
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://***************************************
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (request) SETUP stream 1
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
```

- When I check the CloudWatch metrics in the KVS console, could see that there were no PutMedia events

Not sure if I missed anything during my configuration and would appreciate any help on this.

disa6302, 2020-11-10:
@DManthoko ,

There might be an issue with your pipeline. Please verify that your pipeline works with fakesink. If it does not, then you will have to fix your pipeline

DManthoko, 2020-11-10:
Thanks for taking a look @disa6302. When I run fakesink, this what I see

```
$ gst-launch-1.0 rtspsrc location=rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov short-header=TRUE ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (request) SETUP stream 1
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
```

It just hangs at 'Sent PLAY request'. I don't have much experience with gstreamer, so I do not know what will I need to change in my pipeline. Can you suggest what I should change in my pipeline?

I also have some other questions:

- Is there any good reference on how to use the gstreamer plugin? I tried to follow the parameter reference seen in [1], but  it only shows the default parameters with no examples of how to use them really. 
- Is there an alternative method (using the AWS SDK's perhaps) that I can use to send an RTSP stream to KVS?

[1] GStreamer Element Parameter Reference  - https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin-parameters.html

disa6302, 2020-11-10:
@DManthoko , 

Please try some of the sample pipelines in the repo. One of them you could try is [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md#running-the-gst-launch-10-command-to-start-streaming-from-a-rtsp-camera-source) one.

All examples are available in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md for linux.

You can use the [sample application](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md#running-the-gstreamer-rtsp-sample-application) we provide to see if it works. 

Since your pipeline does not work with `fakesink` as well, this is not an issue with KVS. GStreamer forums would be a good place to look at for some solutions. 

I am closing this for now. Feel free to reach out if you have any questions.

closed 2020-11-10 by disa6302