awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #613: [QUESTION] Failed to allocate heap

Julchenwsf, 2020-11-15:
Hello, I try to build the so file. This works. But when I try to execute it, it says the following error

```

2020-11-15 16:33:58 ERROR   aivHeapInit(): Failed to allocate heap with limit size 1073741824
2020-11-15 16:33:58 ERROR   heapInitialize(): Failed to initialize native heap.
2020-11-15 16:33:58 ERROR   createKinesisVideoClient(): operation returned status code: 0x00000004
2020-11-15 16:33:58 ERROR   semaphoreSetLockInternal(): operation returned status code: 0x00000001
2020-11-15 16:33:58 ERROR   semaphoreWaitUntilClearInternal(): operation returned status code: 0x00000001
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed to create Kinesis Video client. StatusCode: 0x4
        at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:59)
```

Do you have a hint why this happened?

Best regards
Julia

MushMal, 2020-11-16:
@Julchenwsf you are attaching logs for a Java asset to the CPP repository. 

As you can see from the logs it failed to allocate the necessary heap with 1073741824 bytes in size

`2020-11-15 16:33:58 ERROR   aivHeapInit(): Failed to allocate heap with limit size 1073741824
`

I am guessing you are using the Java demo application. You will need to overwrite the default value of the heap size 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/3fc9f3c70e1d8016598d9c4536d19e36cb72499c/src/main/java/com/amazonaws/kinesisvideo/java/client/KinesisVideoJavaClientFactory.java#L34

which is by mistake is named as 64 MB but actually defaults to 1GB.

Or, alternatively, use the other overloaded method in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/demo/com/amazonaws/kinesisvideo/demoapp/DemoAppMain.java#L42

to specify your own version of DeviceInfo with the right size of heap.

Please resolve this issue if you have no further questions regarding this case.

If you have further questions, please cut them in the Java SDK repository.

Julchenwsf, 2020-11-16:
Thanks. This solved the problem.

closed 2020-11-16 by Julchenwsf