awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #615: Not able to stream from RTSP using docker

bhavikapanara, 2020-11-19:
I have successfully created a docker image as defined in this [link](https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/producer-cpp/docker-rtsp). [https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/producer-cpp/docker-rtsp]

successfully created docker image using this command
`$ docker build -t rtspdockertest . `

But when I run the image using below command, I can't see the video in Kinesis Video Stream playback. By observing log messages, it seems that streaming is not working.
Can someone please help me to resolve my query?

`$ docker run -it <IMAGE_ID> <AWS_ACCESS_KEY_ID> <AWS_SECRET_ACCESS_KEY> <RTSP_URL> <STREAM_NAME>`

Here is my log:

```
[INFO ] [19-11-2020 13:22:50:186.636 GMT] Using region: us-west-2
[INFO ] [19-11-2020 13:22:50:186.683 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [19-11-2020 13:22:50:186.696 GMT] No session token was detected.
[INFO ] [19-11-2020 13:22:50:187.622 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [19-11-2020 13:22:50:187.657 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [19-11-2020 13:22:50:187.672 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [19-11-2020 13:22:50:187.700 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [19-11-2020 13:22:50:187.726 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [19-11-2020 13:22:50:187.749 GMT] getSecurityTokenHandler invoked
[DEBUG] [19-11-2020 13:22:50:187.787 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1605792170187771918 Expiration: 0
[INFO ] [19-11-2020 13:22:50:187.801 GMT] New credentials expiration is 1605795770
[DEBUG] [19-11-2020 13:22:50:187.822 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [19-11-2020 13:22:50:187.833 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [19-11-2020 13:22:50:187.850 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [19-11-2020 13:22:50:187.865 GMT] clientReadyHandler invoked
[DEBUG] [19-11-2020 13:22:50:187.896 GMT] Client is ready
[INFO ] [19-11-2020 13:22:50:187.950 GMT] Creating Kinesis Video Stream MilkmanStream_12
[INFO ] [19-11-2020 13:22:50:187.979 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [19-11-2020 13:22:50:187.993 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [19-11-2020 13:22:50:188.005 GMT] logStreamInfo(): 	Stream name: MilkmanStream_12 
[DEBUG] [19-11-2020 13:22:50:188.017 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [19-11-2020 13:22:50:188.028 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [19-11-2020 13:22:50:188.040 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [19-11-2020 13:22:50:188.053 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [19-11-2020 13:22:50:188.064 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [19-11-2020 13:22:50:188.073 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [19-11-2020 13:22:50:188.081 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [19-11-2020 13:22:50:188.089 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [19-11-2020 13:22:50:188.100 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [19-11-2020 13:22:50:188.112 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [19-11-2020 13:22:50:188.125 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [19-11-2020 13:22:50:188.135 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [19-11-2020 13:22:50:188.144 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [19-11-2020 13:22:50:188.155 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [19-11-2020 13:22:50:188.167 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [19-11-2020 13:22:50:188.178 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [19-11-2020 13:22:50:188.191 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [19-11-2020 13:22:50:188.205 GMT] logStreamInfo(): Track list
[DEBUG] [19-11-2020 13:22:50:188.216 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [19-11-2020 13:22:50:188.225 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [19-11-2020 13:22:50:188.233 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [19-11-2020 13:22:50:188.241 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [19-11-2020 13:22:50:188.249 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [19-11-2020 13:22:50:188.302 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [19-11-2020 13:22:51:925.347 GMT] writeHeaderCallback(): RequestId: 30743f16-3566-4c43-98ec-4201f18d8b00
[DEBUG] [19-11-2020 13:22:51:925.723 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.605209058029E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:291596880574:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:291596880574:stream/MilkmanStream_12/1605209058029","StreamName":"MilkmanStream_12","Version":"CLNoJ6pJcaxwnE4mPzmJ"}}
[INFO ] [19-11-2020 13:22:51:928.828 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [19-11-2020 13:22:51:929.030 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [19-11-2020 13:22:53:892.150 GMT] writeHeaderCallback(): RequestId: 62b4503b-8c01-4aaa-926e-07d8c4a49de2
[DEBUG] [19-11-2020 13:22:53:892.471 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [19-11-2020 13:22:53:896.180 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [19-11-2020 13:22:53:896.296 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [19-11-2020 13:22:53:896.356 GMT] getStreamingTokenHandler invoked
[DEBUG] [19-11-2020 13:22:53:896.407 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1605792173896399216 Expiration: 1605795770
[INFO ] [19-11-2020 13:22:53:896.456 GMT] New credentials expiration is 1605795773
[INFO ] [19-11-2020 13:22:53:896.554 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [19-11-2020 13:22:53:896.628 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[... 118 lines cut ...]
[DEBUG] [19-11-2020 13:25:03:149.433 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
[INFO ] [19-11-2020 13:25:03:149.763 GMT] putStreamResultEvent(): Put stream result event. New upload handle 4
[DEBUG] [19-11-2020 13:25:03:160.078 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
[INFO ] [19-11-2020 13:25:04:542.985 GMT] writeHeaderCallback(): RequestId: f6b35392-a0b6-d576-ae83-27e2e634e0b2
[DEBUG] [19-11-2020 13:25:04:543.215 GMT] postReadCallback(): Pausing CURL read for upload handle: 4
[WARN ] [19-11-2020 13:25:34:192.461 GMT] curlCompleteSync(): curl perform failed for url https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [19-11-2020 13:25:34:192.679 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAUHZEHJK7KN7TAHRI/20201119/us-west-2/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=81b9ad9e74a49e895509620353da7a998a2209c2ddf8f1f
[DEBUG] [19-11-2020 13:25:34:192.761 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MilkmanStream_12 with upload handle: 4 exited. http status: 0
[... cut ...]
[DEBUG] [19-11-2020 13:25:46:739.230 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
[DEBUG] [19-11-2020 13:25:46:739.427 GMT] streamReadyHandler invoked
[DEBUG] [19-11-2020 13:25:46:739.500 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
```

And it is going forever.

I would like to request @MushMal to please look at my problem and help me to feed videos from RTSP camera into Kinesis Video Stream.

Thanks a lot,
Bhavika

MushMal, 2020-11-19:
Here are some suggestions:

* Look into your logs and see if there are any errors. 
* Search the issues for the same or similar logs

bhavikapanara, 2020-11-19:
Thanks, @MushMal for your suggestions

I have already search a lot for this error. But no success.
There is no error in the log but here is the warning message:

```
[DEBUG] [19-11-2020 14:56:52:496.285 GMT] postReadCallback(): Pausing CURL read for upload handle: 188
[WARN ] [19-11-2020 14:57:22:625.108 GMT] curlCompleteSync(): curl perform failed for url https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [19-11-2020 14:57:22:625.282 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)

```

MushMal, 2020-11-19:
Search for MAX_FRAGMENT_DURATION_REACHED

bhavikapanara, 2020-11-19:
searched MAX_FRAGMENT_DURATION_REACHED error, but none of the solutions works.

MushMal, 2020-11-19:
Hi there,

Here is the search result: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/search?q=MAX_FRAGMENT_DURATION_REACHED&type=issues

As you can see there are plenty of hits.

Going through them and taking the exact match link: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/537

Please ensure that your pipeline produces valid H264 with up-to 10 seconds GoP. The recommended value is 2-6 seconds.

Refer to your manufacturers guide on how to tune your rtsp producer.

Please resolve this issue if you have no further questions regarding KVS. We won't be able to help you with your media source.

bhavikapanara, 2020-11-22:
Thanks, @MushMal for your valuable feedback.

When I try to stream using GStreamer command, it works perfectly fine. No error. 
I have used below GStreamer command.

`gst-launch-1.0 rtspsrc location="rtsp://admi....." short-header=TRUE ! rtph264depay ! video/x-h264, width=640,height=480,framerate=15/1, format=avc,alignment=au ! kvssink stream-name="Stream_12" storage-size=512 access-key="AKIA........" secret-key="EM++udR7s............" aws-region="us-east-2"
`

But, when I try to stream with docker for RTSP streaming using [this exampe](https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/producer-cpp/docker-rtsp), it gives me an error. It simply indicates that there is no issue with the hardware.

bhavikapanara, 2020-11-22:
@MushMal 

By observing the log message I came to know that When I start streaming by running the following command for docker RTSP, it actually starts to write data into Kinesis Video Stream but after some time it stops. 

Even it able to write data in beginning, I can't see streaming in media playback.
@MushMal Please help me to find the solution.

```
docker run -it <IMAGE_ID> /bin/bash
AWS_ACCESS_KEY_ID=<AWS_ACCESS_KEY_ID> AWS_SECRET_ACCESS_KEY=<AWS_SECRET_ACCESS_KEY> ./kvs_gstreamer_sample <STREAM_NAME> <RTSP_URL>

```
Here is the log message:

```
root@73b557dcce9a:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build# AWS_ACCESS_KEY_ID=AKIAUHZEHJK7KN7TAHRI AWS_SECRET_ACCESS_KEY=EM++udR7sBPD1YKVRhGm7b2RLKtSA4KEN7KYRsqv ./kvs_gstreamer_sample MilkmanStream_12 rtsp://admin:milkman20@26062b.dvrinside.com:554/live12_rec
[INFO ] [22-11-2020 06:35:07:955.263 GMT] Using region: us-west-2
[INFO ] [22-11-2020 06:35:07:955.377 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [22-11-2020 06:35:07:955.402 GMT] No session token was detected.
[INFO ] [22-11-2020 06:35:07:956.573 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [22-11-2020 06:35:07:956.642 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [22-11-2020 06:35:07:956.662 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [22-11-2020 06:35:07:956.693 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [22-11-2020 06:35:07:956.723 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [22-11-2020 06:35:07:956.746 GMT] getSecurityTokenHandler invoked
[DEBUG] [22-11-2020 06:35:07:956.782 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1606026907956767017 Expiration: 0
[INFO ] [22-11-2020 06:35:07:956.803 GMT] New credentials expiration is 1606030507
[DEBUG] [22-11-2020 06:35:07:956.834 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [22-11-2020 06:35:07:956.861 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [22-11-2020 06:35:07:956.882 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [22-11-2020 06:35:07:956.902 GMT] clientReadyHandler invoked
[DEBUG] [22-11-2020 06:35:07:956.937 GMT] Client is ready
[INFO ] [22-11-2020 06:35:07:957.001 GMT] Creating Kinesis Video Stream MilkmanStream_12
[INFO ] [22-11-2020 06:35:07:957.039 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [22-11-2020 06:35:07:957.058 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [22-11-2020 06:35:07:957.073 GMT] logStreamInfo(): 	Stream name: MilkmanStream_12 
[DEBUG] [22-11-2020 06:35:07:957.087 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [22-11-2020 06:35:07:957.102 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [22-11-2020 06:35:07:957.116 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [22-11-2020 06:35:07:957.133 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [22-11-2020 06:35:07:957.148 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [22-11-2020 06:35:07:957.162 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [22-11-2020 06:35:07:957.176 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [22-11-2020 06:35:07:957.190 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [22-11-2020 06:35:07:957.205 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [22-11-2020 06:35:07:957.219 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [22-11-2020 06:35:07:957.234 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [22-11-2020 06:35:07:957.265 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [22-11-2020 06:35:07:957.285 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [22-11-2020 06:35:07:957.300 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [22-11-2020 06:35:07:957.314 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [22-11-2020 06:35:07:957.329 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [22-11-2020 06:35:07:957.343 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [22-11-2020 06:35:07:957.357 GMT] logStreamInfo(): Track list
[DEBUG] [22-11-2020 06:35:07:957.371 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [22-11-2020 06:35:07:957.385 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [22-11-2020 06:35:07:957.400 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [22-11-2020 06:35:07:957.414 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [22-11-2020 06:35:07:957.428 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [22-11-2020 06:35:07:957.491 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [22-11-2020 06:35:08:194.741 GMT] writeHeaderCallback(): RequestId: 440a56d1-c191-4a83-9424-26ba3e898cc8
[DEBUG] [22-11-2020 06:35:08:195.050 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.605209058029E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-west-2:291596880574:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:291596880574:stream/MilkmanStream_12/1605209058029","Strea [... cut ...]
[DEBUG] [22-11-2020 06:36:29:641.522 GMT] postReadCallback(): Pausing CURL read for upload handle: 3
[WARN ] [22-11-2020 06:36:59:642.576 GMT] curlCompleteSync(): curl perform failed for url https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [22-11-2020 06:36:59:642.786 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com/putMedia
Request Headers:

```

MushMal, 2020-11-22:
To reflect back. You have the exact same gst-launch-1.0 command line that you are executing from the command line of your computer and it works but running the exact same command line from within the docket instance produces MAX_FRAGMENT_DURATION_REACHED?

If so I can't give any rational explanation with the exception of

* Your GoP size is close to 10 seconds and you are just running over the limit
* You have a lot of packet drop in the docker and you are dropping I-frames which then produces large GoP.

Recommend you running a few experiments and ensure you have solid data points. If the issue was really what I reflected back then try to configure your rtsp source to produce smaller GoP size. Check for packet loss between your rtsp source and your computer.

Your logs clearly indicate:

[DEBUG] [22-11-2020 06:35:25:934.928 GMT] postWriteCallback(): Curl post body write function for stream with handle: MilkmanStream_12 and upload handle: 0 returned: {"EventType":"ERROR","FragmentTimecode":1606026907955,"FragmentNumber":"91343852333181739401811805407331952656150960289","ErrorCode":"MAX_FRAGMENT_DURATION_REACHED","ErrorId":4002}

MushMal, 2020-11-22:
A couple of other observations. On the retry, you are hitting timeouts which is an indicator of a bad network.

Please work on sorting out the network issues too. Try selecting a region closest to your physical location.

bhavikapanara, 2020-11-22:
I have also checked that but no success.

frequently I get this error:

1.
`returned: {"EventType":"ERROR","FragmentTimecode":1606035243121,"FragmentNumber":"91343852333181749305332119690376393448284629848","ErrorCode":"MAX_FRAGMENT_DURATION_REACHED","ErrorId":4002}
`

2.
```
curlCompleteSync(): curl perform failed for url https://s-8451b5cd.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [22-11-2020 08:55:04:255.319 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)

```

Do I need to change any default parameter value in file [kvs_gstreamer_sample.cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp)? 

```
void create_kinesis_video_frame(Frame *frame, const nanoseconds &pts, const nanoseconds &dts, FRAME_FLAGS flags,
                                void *data, size_t len) {
    frame->flags = flags;
    frame->decodingTs = static_cast<UINT64>(dts.count()) / DEFAULT_TIME_UNIT_IN_NANOS;
    frame->presentationTs = static_cast<UINT64>(pts.count()) / DEFAULT_TIME_UNIT_IN_NANOS;
    // set duration to 0 due to potential high spew from rtsp streams
    frame->duration = 0;
    frame->size = static_cast<UINT32>(len);
    frame->frameData = reinterpret_cast<PBYTE>(data);
    frame->trackId = DEFAULT_TRACK_ID;
}
```

For one of your comment in similar[ issue](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/33), you suggest updating the above paragraph to this

```
void create_kinesis_video_frame(Frame *frame, const nanoseconds &pts, const nanoseconds &dts, FRAME_FLAGS flags,
void *data, size_t len) {
frame->flags = flags;
frame->decodingTs = static_cast(dts.count()) / DEFAULT_TIME_UNIT_IN_NANOS;
frame->presentationTs = static_cast(pts.count()) / DEFAULT_TIME_UNIT_IN_NANOS;
frame->duration = 20 * HUNDREDS_OF_NANOS_IN_A_MILLISECOND;
frame->size = static_cast(len);
frame->frameData = reinterpret_cast(data);
}

```

@MushMal Can you please look at this the file **kvs_gstreamer_sample.cpp**?  If there is any change help me to resolve the issue.

MushMal, 2020-11-22:
You have a networking issue and a max fragment duration issue.  One are actionable from the sdk. Per my suggestions earlier, you need to tune your encoder in the rtsp side and ensure good networks throughput.

You also have not replied to any of my comments about your usage and whether you see any of the issues with the stock samples eliminating the rtsp.

At this stage there is nothing I can suggest as the data you provided simply doesn’t make sense when you mentioned that the same rtsp source works ok but with the docker image it produces these logs.

Please take some time and compile your findings properly so you can trust your data and your logs.
We have been iterating over these issues multiple times with no further information or data that was requested.

bhavikapanara, 2020-11-22:
> To reflect back. You have the exact same gst-launch-1.0 command line that you are executing from the command line of your computer and it works but running the exact same command line from within [...]

For docker, I am not using the GStreamer command. Instead, I am following this tutorial [https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/producer-cpp/docker-rtsp](https://github.com/aws-samples/amazon-kinesis-video-streams-demos/tree/master/producer-cpp/docker-rtsp) to run docker for RTSP streaming using c++ producer.

disa6302, 2020-11-23:
@bhavikapanara , 

Check the gstreamer sample app. It is possible you will have to modify some gstreamer params there to get it working with your rtsp camera. If yes, then the docker image would not work as it is. You might have to build the docker image with any changes you might make in the application. 

Ensure your gstreamer sample app works as is on your machine.

closed 2020-12-05 by bhavikapanara