awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #616: [BUG] Got "Caught SIGSEGV" error after sending kvs-add-metadata event to kvssink

phongvh, 2020-11-20:
Hi,
I created a gstreamer plugin call "myfilter" to use as a filter before kvssink. The only thing this plugin does is sending a CUSTOM_DOWNSTREAM event with name "kvs-add-metadata" to kvssink to set metadata info to the fragments like this:

`gst_pad_push_event(filter->srcpad, gst_event_new_custom(GST_EVENT_CUSTOM_DOWNSTREAM, gst_structure_new("kvs-add-metadata", "name", "string", "author", "value", "string", "Phong", "persist", "boolean", "true", NULL)));`

The gstreamer pipeline I used is this:

`gst-launch-1.0 v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,format=I420,width=1920,height=1080 ! x264enc bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! **myfilter** ! kvssink name=sink stream-name="LiveStreaming" storage-size=512 access-key="[SECRET]" secret-key="[SECRET]" aws-region="us-west-2" pulsesrc ! audioconvert ! avenc_aac ! queue ! sink.`

After run that I got "Caught SIGSEGV" error.

I'm not sure if it's a bug from kvssink plugin or I did something wrong here. Pls give me some advice. Thanks!

MushMal, 2020-11-20:
@phongvh there is absolutely nothing actionable in the issue. 

* Try to build a debug version of the sink
* Run with GST_DEBUG=5 env var set
* Run with VERBOSE logging enabled in the kvs
* Run it under ldd to catch the crash
* Analyze the crash which would give you some idea what's going on
* Attach debug logs and the crash stack with symbol information

In case if the crash originates from the kvssink, try to reproduce it so we can look into it.

Removing BUG tag as we are not sure yet what causes this crash and whether it's KVS related.

phongvh, 2020-11-23:
@MushMal Thanks for your reply! I ran this command 
`LD_DEBUG=all GST_DEBUG=5 gst-launch-1.0 v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,format=I420,width=1920,height=1080 ! x264enc bframes=0  key-int-max=45 bitrate=512 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! myfilter ! kvssink name=sink stream-name="LiveStreaming" storage-size=512 access-key="[deleted]" secret-key="[deleted]" aws-region="us-west-2" `
and I got this log. 
[gst_debug.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5581514/gst_debug.log)

And this is for ld debug
[ld.debug.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/5581483/ld.debug.zip)

I looked through the log but still don't know what causes the problem.

Here is the code of the plugin gstmyfilter.cpp
`/**
 * SECTION:element-myfilter
 *
 * FIXME:Describe myfilter here.
 *
 * <refsect2>
 * <title>Example launch line</title>
 * |[
 * gst-launch -v -m fakesrc ! myfilter ! fakesink silent=TRUE
 * ]|
 * </refsect2>
 */

#ifdef HAVE_CONFIG_H
#  include <config.h>
#endif

#include <gst/gst.h>
#include <thread>
#include "gstmyfilter.h"

GST_DEBUG_CATEGORY_STATIC (gst_my_filter_debug);
#define GST_CAT_DEFAULT gst_my_filter_debug

/* Filter signals and args */
enum
{
  /* FILL ME */
  LAST_SIGNAL
};

enum
{
  PROP_0,
  PROP_SILENT
};

/* the capabilities of the inputs and outputs.
 *
 * describe the real formats here.
 */
static GstStaticPadTemplate sink_factory = GST_STATIC_PAD_TEMPLATE ("sink",
    GST_PAD_SINK,
    GST_PAD_ALWAYS,
    GST_STATIC_CAPS ("ANY")
    );

static GstStaticPadTemplate src_factory = GST_STATIC_PAD_TEMPLATE ("src",
    GST_PAD_SRC,
    GST_PAD_ALWAYS,
    GST_STATIC_CAPS ("ANY")
    );

#define gst_my_filter_parent_class parent_class
G_DEFINE_TYPE (GstMyFilter, gst_my_filter, GST_TYPE_ELEMENT);

static void gst_my_filter_set_property (GObject * object, guint prop_id,
    const GValue * value, GParamSpec * pspec);
static void gst_my_filter_get_property (GObject * object, guint prop_id,
[... 214 lines cut ...]

/* PACKAGE: this is usually set by meson depending on some _INIT macro
 * in meson.build and then written into and defined in config.h, but we can
 * just set it ourselves here in case someone doesn't use meson to
 * compile this code. GST_PLUGIN_DEFINE needs PACKAGE to be defined.
 */
#ifndef PACKAGE
#define PACKAGE "myfirstmyfilter"
#endif

/* gstreamer looks for this structure to register myfilters
 *
 * exchange the string 'Template myfilter' with your myfilter description
 */
GST_PLUGIN_DEFINE (
    GST_VERSION_MAJOR,
    GST_VERSION_MINOR,
    myfilter,
    "Template myfilter",
    myfilter_init,
     "1.0",
        "Proprietary",
        "GStreamer",
        "http://gstreamer.net/"
)`

MushMal, 2020-11-23:
Sorry, I am not sure what I am looking at and why. I don't see any crash stacks with symbols. I haven't looked at your code.

Here are some suggestions on top of what I already suggested but didn't get any replies

* Start with a known platform on a commodity hardware. For example Linux on x64 platform or Mac
* Get the included samples working.
* Get your media pipeline in GStreamer working WITHOUT any uses of KVS first.

These are important steps for you to debug.

Please resolve this issue if you have nor further questions related to KVS itself.

phongvh, 2020-11-24:
@MushMal for the ld log, I set the env LD_DEBUG_OUTPUT=ld.debug, so it save the ldd log to the separate files. I didn't print it together with GST_DEBUG since the log would be too long.

My machine is Ubuntu 18.04. I've already streamed the video/audio successfully to Amazon before, using the same command without the plugin myfilter. Like this 
`gst-launch-1.0 v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,format=I420,width=1920,height=1080 ! x264enc bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name="LiveStreaming" storage-size=512 access-key="[SECRET]" secret-key="[SECRET]" aws-region="us-west-2" pulsesrc ! audioconvert ! avenc_aac ! queue ! sink.`

Or if I add the myfilter plugin without the command `gst_pad_push_event(filter->srcpad, gst_event_new_custom(GST_EVENT_CUSTOM_DOWNSTREAM, gst_structure_new("kvs-add-metadata", "name", "string", "author", "value", "string", "Phong", "persist", "boolean", "true", NULL)));`, it still works well. The issue only happens when I add the command into myfilter plugin.

My ultimate purpose is to attach some custom metadatas together with the video/audio stream. When I read the source code of kvssink, I saw that kvssink allows to add metadata to the fragment in gst_kvs_sink_handle_sink_event function 
`case GST_EVENT_CUSTOM_DOWNSTREAM: {
            const GstStructure *structure = gst_event_get_structure(event);
            std::string metadata_name, metadata_value;
            gboolean persistent;
            bool is_persist;

            if (!gst_structure_has_name(structure, KVS_ADD_METADATA_G_STRUCT_NAME)) {
                goto CleanUp;
            }

            LOG_INFO("received kvs-add-metadata event");
            if (NULL == gst_structure_get_string(structure, KVS_ADD_METADATA_NAME) ||
                NULL == gst_structure_get_string(structure, KVS_ADD_METADATA_VALUE) ||
                FALSE == gst_structure_get_boolean(structure, KVS_ADD_METADATA_PERSISTENT, &persistent)) {

                LOG_WARN("Event structure contains invalid field: " << std::string(gst_structure_to_string (structure)));
                goto CleanUp;
            }

            metadata_name = std::string(gst_structure_get_string(structure, KVS_ADD_METADATA_NAME));
            metadata_value = std::string(gst_structure_get_string(structure, KVS_ADD_METADATA_VALUE));
            is_persist = persistent;

            bool result = data->kinesis_video_stream->putFragmentMetadata(metadata_name, metadata_value, is_persist);
            if (false == result) {
                LOG_WARN("Failed to putFragmentMetadata. name: " << metadata_name << ", value: " << metadata_value << ", persistent: " << is_persist);
            }
            gst_event_unref (event);
            event = NULL;
            break;
        }`
That's why I want to write a myfilter plugin to send a kvs-add-metadata event to kvssink.

Could you show me how to do it in the right way? I searched and haven't found any document on that.

MushMal, 2020-11-24:
I am not following... you don’t have a repro step and you don’t even have a crash stack. I won’t be able to debug your application. Can you simply start with the default sample application to ensure it works for you. Try to modify it as you go. At this stage I don’t know where your crash is.

Try to debug your application further.

disa6302, 2020-12-07:
Closing since it has gone stale.

closed 2020-12-07 by disa6302

rtylecek, 2021-08-02:
>  Could you show me how to do it in the right way? I searched and haven't found any document on that.

@MushMal There is still no specific documentation for the task of attaching metadata in gstkvssink. 
I could only find these:

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-meta.html

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/f68a50697436e8d1285ef297d5817b112841cb73/docs/structures.md#fragment-metadata

This is just superficial and not sufficient to implement the solution.

MushMal, 2021-08-02:
This is how the custom event is handled in the plugin:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L970

You might want to follow up with the GStreamer forums to see why you are unable to send the message and what's crashing. I highly recommend constructing your pipeline without the KVS sink and instead use fakesink to stabilize your application first. Check out the GStreamer docs on custom events: https://gstreamer.freedesktop.org/documentation/gstreamer/gstevent.html?gi-language=c#gst_event_new_custom