awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #617: [BUG] Unable to stream to KVS from Pi from HTTP stream source

TriHoangSG, 2020-11-22:
I'm unable to stream HTTP stream from an IP camera to KVS from a Raspberry.

Device: Raspberry 0010 - Model B+ with 512M
KVS stream producer docker image: 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi:latest
Command use: GST_DEBUG=4 AWS_ACCESS_KEY_ID=AAA AWS_SECRET_ACCESS_KEY=BBB ./kvs_gstreamer_sample TEST http://user:pass@192.168.0.201/video.cgi

```
DeviceResultEvent(): Create device result event.
[DEBUG] [22-11-2020 17:02:00:052.634 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [22-11-2020 17:02:00:053.117 GMT] clientReadyHandler invoked
[DEBUG] [22-11-2020 17:02:00:056.041 GMT] Client is ready
[INFO ] [22-11-2020 17:02:00:057.788 GMT] Creating Kinesis Video Stream TEST
[INFO ] [22-11-2020 17:02:00:059.183 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [22-11-2020 17:02:00:062.616 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [22-11-2020 17:02:00:063.035 GMT] logStreamInfo():      Stream name: TEST
[DEBUG] [22-11-2020 17:02:00:063.481 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [22-11-2020 17:02:00:063.927 GMT] logStreamInfo():      Content type: video/h264
[DEBUG] [22-11-2020 17:02:00:065.047 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [22-11-2020 17:02:00:068.604 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [22-11-2020 17:02:00:069.071 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [22-11-2020 17:02:00:071.394 GMT] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [22-11-2020 17:02:00:071.854 GMT] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [22-11-2020 17:02:00:074.392 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [22-11-2020 17:02:00:074.887 GMT] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [22-11-2020 17:02:00:075.342 GMT] logStreamInfo():      Framerate: 25
[DEBUG] [22-11-2020 17:02:00:076.290 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [22-11-2020 17:02:00:076.753 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [22-11-2020 17:02:00:077.210 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [22-11-2020 17:02:00:077.658 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [22-11-2020 17:02:00:079.015 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [22-11-2020 17:02:00:079.475 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [22-11-2020 17:02:00:082.389 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [22-11-2020 17:02:00:083.237 GMT] logStreamInfo(): Track list
[DEBUG] [22-11-2020 17:02:00:083.623 GMT] logStreamInfo():      Track id: 1
[DEBUG] [22-11-2020 17:02:00:084.062 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [22-11-2020 17:02:00:085.220 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [22-11-2020 17:02:00:085.676 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [22-11-2020 17:02:00:088.734 GMT] logStreamInfo():      Track cpd: NULL
[DEBUG] [22-11-2020 17:02:00:090.229 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [22-11-2020 17:02:01:332.737 GMT] writeHeaderCallback(): RequestId: 940e2845-3f99-4c6d-9f23-beeb7fd62e1c
[DEBUG] [22-11-2020 17:02:01:336.476 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.606056377417E9,"DataRetentionInHours":24,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-southeast-1:322679545901:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-southeast-1:322679545901:stream/TEST/1606056377417","StreamName":"TEST","Version":"GZIbG2zhec63IHkeVCZl"}}
[INFO ] [22-11-2020 17:02:01:358.416 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [22-11-2020 17:02:01:363.348 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [22-11-2020 17:02:01:367.054 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [22-11-2020 17:02:01:367.497 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [22-11-2020 17:02:02:548.558 GMT] writeHeaderCallback(): RequestId: 6d0dd964-ec9b-4618-82d0-e301f11256f7
[DEBUG] [22-11-2020 17:02:02:557.179 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-670061c1.kinesisvideo.ap-southeast-1.amazonaws.com"}
[INFO ] [22-11-2020 17:02:02:591.804 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [22-11-2020 17:02:02:599.852 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [22-11-2020 17:02:02:602.411 GMT] getStreamingTokenHandler invoked
[DEBUG] [22-11-2020 17:02:02:603.642 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1606064522603510336 Expiration: 1606068120
[INFO ] [22-11-2020 17:02:02:604.347 GMT] New credentials expiration is 1606068122
[INFO ] [22-11-2020 17:02:02:607.689 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [22-11-2020 17:02:02:608.681 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [22-11-2020 17:02:02:616.038 GMT] streamReadyHandler invoked
[DEBUG] [22-11-2020 17:02:02:617.976 GMT] Stream is ready
0:00:00.001238991    91   0x3bf000 INFO                GST_INIT gst.c:510:init_pre: Initializing GStreamer Core Library version 1.10.4
0:00:00.006423954    91   0x3bf000 INFO                GST_INIT gst.c:511:init_pre: Using library installed in /usr/lib/arm-linux-gnueabihf
0:00:00.006744952    91   0x3bf000 INFO                GST_INIT gst.c:522:init_pre: Linux 51b14c30f05e 5.4.72+ #1356 Thu Oct 22 13:56:00 BST 2020 armv6l
0:00:00.009554932    91   0x3bf000 INFO                GST_INIT gstmessage.c:126:_priv_gst_message_initialize: init messages
0:00:00.020975852    91   0x3bf000 INFO                GST_INIT gstcontext.c:83:_priv_gst_context_initialize: init contexts
0:00:00.030341786    91   0x3bf000 INFO      GST_PLUGIN_LOADING gstplugin.c:316:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.036463743    91   0x3bf000 INFO      GST_PLUGIN_LOADING gstplugin.c:224:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.041125710    91   0x3bf000 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.041647706    91   0x3bf000 INFO            GST_REGISTRY gstregistry.c:1738:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.234702346    91   0x3bf000 INFO            GST_REGISTRY gstregistrybinary.c:619:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.arm.bin in 0.192569 seconds
0:00:00.239885310    91   0x3bf000 INFO            GST_REGISTRY gstregistry.c:1594:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.292339940    91   0x3bf000 INFO            GST_REGISTRY gstregistry.c:1696:scan_and_update_registry: Registry cache has not changed
0:00:00.296367912    91   0x3bf000 INFO            GST_REGISTRY gstregistry.c:1773:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.299843887    91   0x3bf000 INFO                GST_INIT gst.c:720:init_post: GLib runtime version: 2.50.3
[... 21 lines cut ...]
0:00:00.468519699    91   0x3bf000 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstomx-rpi.so" loaded
0:00:00.473503664    91   0x3bf000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:362:gst_element_factory_create: creating element "omxh264enc" named "encoder"
0:00:00.476399644    91   0x3bf000 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x4c00b8> adding pad 'sink'
0:00:00.479752620    91   0x3bf000 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x4c00b8> adding pad 'src'
[... cut ...]
[INFO ] [22-11-2020 17:02:03:608.399 GMT] Freeing Kinesis Video Stream TEST
[INFO ] [22-11-2020 17:02:03:609.038 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [22-11-2020 17:02:03:609.408 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [22-11-2020 17:02:04:011.468 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [22-11-2020 17:02:04:012.111 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

TriHoangSG, 2020-11-22:
From inside container, I can do wget on http://user:pass@192.168.0.201/video.cgi and get video feed as file without issue.

MushMal, 2020-11-22:
The logs clearly show that buoy are unable to open the camera device. You can search the RPi forums and the issues in KVS cpp repository for more info.

As is, this is not KVS sdk related issue. Removing tags

disa6302, 2020-11-22:
Closing the issue.

closed 2020-11-22 by disa6302

TriHoangSG, 2020-11-23:
The producer repo has issue, refer here https://dev.to/rohansawant/the-story-of-how-i-kinda-fixed-aws-kinesis-producers-helped-a-student-and-saved-humanity-hundreds-of-man-hours-17e8