awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #618: [BUG] Producer repo issue, required checkout of old commit to get it working

TriHoangSG, 2020-11-23:
Required to checkout commit 824de900c2d4b42a403319ace5de9d68ec88b171 to get Producer working.

See for more details https://dev.to/rohansawant/the-story-of-how-i-kinda-fixed-aws-kinesis-producers-helped-a-student-and-saved-humanity-hundreds-of-man-hours-17e8

hassanctech, 2020-11-23:
Thanks for bringing this to our attention.  We are aware of existing sync issues between the GitHub docs and the AWS documentation and are working to get those resolved.  For the benefit of other readers I will note that the commit mentioned above itself is not the issue, I believe it was only chosen because at that time we actually used to have an install script (to which the docs allude).  The decision was made some time back to remove the install script and to have the dependences as submodules and use cmake to pull down whatever is necessary.  I have not tried on an Raspberry Pi with the latest version but we will investigate these issues, unfortunately I do not have a time line right now when we'll be able to correct the guidance for Raspberry Pi for the most recent release of the SDK.

disa6302, 2020-12-07:
@VenomrushCT ,

Closing this in favor of this: #628 . We will fix this ASAP.

closed 2020-12-07 by disa6302