awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #619: [QUESTION]Error codes on describe stream

Subburaj47, 2020-11-23:
During the application playback, the following error was logged.
`[ERROR][2020-10-17 14:25:31] describeStreamResultEvent(): operation returned status code: 0x52000011`

After this, we have a logic to uninit and recreate the stream from application. While recreation, the following error codes occur in a stretch (Which includes "stream duplicate" error code too).
`201017-14:25:31.855116 [mod=CVRUPLOAD, lvl=INFO] [tid=3849] upload/kvsupload.cpp(690): content_type = video/h264,audio/aac
[ERROR][2020-10-17 14:25:31] createKinesisVideoStream(): operation returned status code: 0x1500001e
[ERROR][2020-10-17 14:25:31] createKinesisVideoStreamSync(): operation returned status code: 0x1500001e
[ERROR][2020-10-17 14:25:31]  Unable to create Kinesis Video stream. 7812807528219129332 Error status: 0x1500001e
201017-14:25:31.907735 [mod=CVRUPLOAD, lvl=ERROR] [tid=3849] upload/kvsupload.cpp(769) : Failed to create kinesis video stream : retrying 
201017-14:25:33.924098 [mod=CVRUPLOAD, lvl=INFO] [tid=3849] upload/kvsupload.cpp(690): content_type = video/h264,audio/aac
[ERROR][2020-10-17 14:25:33] createKinesisVideoStream(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:33] createKinesisVideoStreamSync(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:33]  Unable to create Kinesis Video stream. 7812807528219129332 Error status: 0x5200004b
201017-14:25:33.925370 [mod=CVRUPLOAD, lvl=ERROR] [tid=3849] upload/kvsupload.cpp(769) : Failed to create kinesis video stream : retrying 
201017-14:25:35.925775 [mod=CVRUPLOAD, lvl=INFO] [tid=3849] upload/kvsupload.cpp(690): content_type = video/h264,audio/aac
[ERROR][2020-10-17 14:25:35] createKinesisVideoStream(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:35] createKinesisVideoStreamSync(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:35]  Unable to create Kinesis Video stream. 7812807528219129332 Error status: 0x5200004b
201017-14:25:35.927051 [mod=CVRUPLOAD, lvl=ERROR] [tid=3849] upload/kvsupload.cpp(769) : Failed to create kinesis video stream : retrying 
201017-14:25:37.927420 [mod=CVRUPLOAD, lvl=INFO] [tid=3849] upload/kvsupload.cpp(690): content_type = video/h264,audio/aac
[ERROR][2020-10-17 14:25:37] createKinesisVideoStream(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:37] createKinesisVideoStreamSync(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:37]  Unable to create Kinesis Video stream. 7812807528219129332 Error status: 0x5200004b
201017-14:25:37.928574 [mod=CVRUPLOAD, lvl=ERROR] [tid=3849] upload/kvsupload.cpp(769) : Failed to create kinesis video stream : retrying 
[ERROR][2020-10-17 14:25:38] describeStreamResultEvent(): operation returned status code: 0x52000011
201017-14:25:39.928985 [mod=CVRUPLOAD, lvl=INFO] [tid=3849] upload/kvsupload.cpp(690): content_type = video/h264,audio/aac
[ERROR][2020-10-17 14:25:39] createKinesisVideoStream(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:39] createKinesisVideoStreamSync(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:39]  Unable to create Kinesis Video stream. 7812807528219129332 Error status: 0x5200004b
201017-14:25:39.930268 [mod=CVRUPLOAD, lvl=ERROR] [tid=3849] upload/kvsupload.cpp(769) : Failed to create kinesis video stream : retrying 
201017-14:25:41.930652 [mod=CVRUPLOAD, lvl=INFO] [tid=3849] upload/kvsupload.cpp(690): content_type = video/h264,audio/aac
[ERROR][2020-10-17 14:25:41] createKinesisVideoStream(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:41] createKinesisVideoStreamSync(): operation returned status code: 0x5200004b
[ERROR][2020-10-17 14:25:41]  Unable to create Kinesis Video stream. 7812807528219129332 Error status: 0x5200004b
201017-14:25:41.931853 [mod=CVRUPLOAD, lvl=ERROR] [tid=3849] upload/kvsupload.cpp(769) : Failed to create kinesis video stream : retrying 
201017-14:25:43.932277 [mod=CVRUPLOAD, lvl=ERROR] [tid=3849] upload/kvsupload.cpp(774) : FATAL : Max retry reached in recreate_stream exit process 2128867148
201017-14:25:43.932558 [mod=CVR, lvl=INFO] [tid=3849] core/cvr_daemon.cpp(1686): FATAL : Max retry reached in Re-Invoking kvsStreamInit exit process ` 

Could someone provide clarity on 
- When this scenario could occur
- Is there anything to be additionally added in the uninit to handle such errors?
 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

disa6302, 2020-11-23:
@Subburaj47 , 

I notice that curl init fails. I am unsure why. Also, can you detail on what you do in uninit?

disa6302, 2020-11-23:
Also, one thing I can think of for curl init failing is `curl_global_init` is not thread safe. Is it possible in your application that maybe multiple libraries in your application are using libcurl at the same time?

MushMal, 2020-11-23:
The root cause of the issue is

STATUS_MAX_IOT_THING_NAME_LENGTH

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/src/include/com/amazonaws/kinesis/video/common/Include.h#L41

Please fix this and it should work OK - most likely a not NULL terminated parameter passed into IoT credential provider.

0x5200004b is also an indication that your application attempts to create a stream with one that already is defined. This is a client side issue only. Please track and fix this too.

A higher level status code of 0x52000011 is returned after the state machine exhausted the retry counts.

Let us know if you have KVS related questions and if not please go ahead and resolve.

disa6302, 2020-12-01:
Closing based on tag.

closed 2020-12-01 by disa6302