awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #620: [QUESTION] What is the correct way of adding fragment metadata?

aaronsaifengliu, 2020-11-25:
We are trying to add a set of persistent metadata to each fragment, by modifying the **kvs_gstreamer_sample.cpp** code. We are expecting to see metadata appended to each fragment, but after consuming the stream with the consumer library, we found that metadata were added the first fragment only.  

Specifically, before the "gstreamer_init(argc, argv, &data)" part in "kvs_gstreamer_sample", I added the following line:
data.kinesis_video_stream->putFragmentMetadata(metadata_key, metadata_val, true);

Is this the correct way of adding persistent metadata to all the fragments? 

Thank you very much for your help!

hassanctech, 2020-11-25:
We provide a callback for this in the `kvssink` plugin, please see:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L954

If you run `gst-inspect-1.0 kvssink` you will see the available events to which you can attach a callback function.  Let me know if you need more help how to do that, unfortunately it looks like none of our samples illustrate this, but it is a standard gstreamer element event call back.

aaronsaifengliu, 2020-11-25:
Is it possible for you to provide a sample function for adding fragment metadata, or an example on how to use the callback in kvssink to add fragment metadata? This will be very helpful for anyone who is not familiar with gstreamer but are still interested in using kinesis video stream. If not, I will close this issue for now.

MushMal, 2020-11-26:
@aaronsaifengliu we will consider adding the call to the sample GStreamer application at its priority order.

For now, check out the GStreamer page discussing on how to produce custom GStreamer events here: https://gstreamer.freedesktop.org/documentation/gstreamer/gstevent.html?gi-language=c

Please resolve this issue if you don't have any KVS related/specific questions.

closed 2020-11-27 by aaronsaifengliu