awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #622: [QUESTION] AXIS P3354 Fixed Dome Network Camera with KVS configuration

bml1g12, 2020-12-01:
I see that here: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/system-requirements.html 

Kinesis video streams has been tested with "AXIS P3354 Fixed Dome Network Camera". I am looking to setup Kinesis SDK on an AXIS encoder (P7304) but am looking for resources that could help me figure out what steps would be needed here. It would be very helpful if you could share the details of how you setup Kinesis video streams with the AXIS P3354 - thank you!

Kind Regards,

dfcarpenter, 2020-12-01:
My company uses Axis cameras as well but unfortunately there is no "onboard" integration directly to AWS Kinesis Video streams like there is for [Vivotek cameras](https://www.vivotek.com/amazon-kinesis-video-streams) ( and a few others ). Hence, you will either need to write this integration yourself and load it onto the camera using [ACAP](https://www.axis.com/support/developer-support/axis-camera-application-platform). Here are some [github examples](https://github.com/AxisCommunications/acap3-examples).  We are in the process of creating an integration but it will not be publicly available unfortunately. Your second option is to run the producer from a system on the same network as the camera producing the feed. I would suggest a server running docker and you can run the [producer in a container](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker). The producer will then capture the stream and shuffle it up to AWS. You may run into an issue of dealing with passwords. A quick and dirty fix is to clone the c++ sdk here and edit the [kvs gstreamer file](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L881-L884) and modify the rtspsrc to include a hardcoded username and password. This is mentioned here #501 
```
g_object_set(G_OBJECT (source),
                 "location", data->rtsp_url.c_str(),
                 "short-header", true, // Necessary for target camera
                 "user-id", "",
                 "user-pw", "",
                 NULL);
```
This will be easiest if you have the necessary software to compile the library and then include it in a docker container. Hopefully this helps at least a little

bml1g12, 2020-12-02:
> My company uses Axis cameras as well but unfortunately there is no "onboard" integration directly to AWS Kinesis Video streams like there is for [Vivotek cameras](https://www.vivotek.com/amazon-kinesis-video-streams) ( and a few others ). Hence, you will either need to write this integration yourself and load it onto the camera using [ACAP](https://www.axis.com/support/developer-support/axis-camera-application-platform). Here are some [github examples](https://github.com/AxisCommunications/acap3-examples). We are in the process of creating an integration but it will not be publicly available unfortunately. Your second option is to run the producer from a system on the same network as the camera producing the feed. I would suggest a server running docker and you can run the [producer in a container](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker). The producer will then capture the stream and shuffle it up to AWS. You may run into an issue of dealing with passwords. A quick and dirty fix is to clone the c++ sdk here and edit the [kvs gstreamer file](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L881-L884) and modify the rtspsrc to include a hardcoded username and password. This is mentioned here #501
> 
> ```
> g_object_set(G_OBJECT (source),
>                  "location", data->rtsp_url.c_str(),
>                  "short-header", true, // Necessary for target camera
>                  "user-id", "",
>                  "user-pw", "",
>                  NULL);
> ```
> 
> The url for an axis camera will end up looking like this.
> `rtsp://<username>:<password>@<ip-address>/axis-media/media.amp`
> This will be easiest if you have the necessary software to compile the library and then include it in a docker container. Hopefully this helps at least a little

Thanks that's very helpful. I'd be interested to hear how your company gets on with writing the integration, our company is discussing whether its worth investing the time in this, but its hard to gauge how many man-hours it might consume. There might be some scope for collaboration between our companies if we were to decide Kinesis is the way forward. 

Currently we are using the SMB encrypted file transfer to send video from the camera to the cloud; but I can imagine Kinesis would be more easy to scale to thousands of cameras. We are also investigating HTTPS for video clip transfers but even the AXIS support seems to have little idea how this works.  We can access the RTMP stream from a local device and could take this approach if needed; but one our goals is to remove local hardware where possible, so installing Kinesis SDK on the device itself seems ideal.

MushMal, 2020-12-02:
Great discussion here. The main issue with the device integration is the platform openness. I am not really familiar with this platform and it’s capabilities but I would assume it runs a sort of embedded Linux. The integrator only needs access to the media pipeline as the rest is generic. If the device supports gstreamer then there is a plugin that the integrator could use.

Unfortunately, KVS team does not own the platform so we support generic media pipeline which is GStreamer

dfcarpenter, 2020-12-02:
@bml1g12 One thing I would strongly recommend is to try out using the producer through docker and understand the end to end flow as much as possible. There are many settings to tweak and it can make a difference in cost/overhead/maintenance. If you already have well defined processes around processing for use cases like ML and associated tasks like retention than great but I would at least get to a level of experience with the client and it's settings where you can do "napkin math" and estimate properly based on your end state. At the end of the day it doesn't really matter whether the producer is on the cameras or not but it does make rollout and maintenance somewhat easier.

bml1g12, 2020-12-02:
> Unfortunately, KVS team does not own the platform so we support generic media pipeline which is GStreamer

I understand, but it sounds like the KVS team have already tested it on AXIS P3354 (at least based on https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/system-requirements.html) so I wonder if it would be possible to share the code used for this; even if its not officially supported. 

> @bml1g12 One thing I would strongly recommend is to try out using the producer through docker and understand the end to end flow as much as possible. There are many settings to tweak and it can make [...]

Thanks for the advice. We want to avoid installing servers into the network aside the cameras themselves; so if we were to put the producer elsewhere it would need to be outside of the network (e.g. an EC2 instance) - but then this would then presumably means we would need to try and tunnel through the firewall surrounding the cameras; which would be problematic as we are likely not permitted to allow inbound traffic on the router settings for security reasons.

hassanctech, 2020-12-08:
@bml1g12 Yes I can help you with that!  There are a couple different approaches you can take, as suggested by @dfcarpenter you can become part of Axis' ACAP program and have access to their embedded SDK (and their provided toolchain) so you get access to the media pipeline and can feed frames to our system.  

There is another way.  Most Axis camera's I'm aware of have a proper gstreamer installation.  You can verify this by sshing onto your camera (or in this case your encoder) and at the terminal type: 

`gst-inspect-1.0 --version`

If gst-inspect-1.0 is on the machine (should already be in path) it will print out the version.  Now you can build our kvssink plugin and as a test you can provide a pipeline via gst-launch-1.0 directly on the command line sending video frames to KVS (not your production solution but it's a good test to verify things are working).

The work is all in having the toolchain set up and building pre req's for kvssink and then the plugin itself.  I have tested with the following toolchain:  https://packages.ubuntu.com/xenial/g++-5-arm-linux-gnueabihf

Yes you'll need an Ubuntu 16.04 box for this (I tested on an EC2 instance) to use this toolchain, a newer compiler *should* work fine too, but make sure its "arm-linux-gnueabihf", an on an Ubuntu box.  If you need more assistance on this part let me know.

Once you have your toolchain set up, the proceed with building ssl/curl and other dependences.   Note the `kvssink` plugin is useful to test with the gstreamer installation already on the camera on the command line, but then once you've verified it works you can use ACAP to actually create an application which build the same pipeline (in C), Axis provided source plugin is `artpecsrc` which already goes through the hardware encoder and produces an h264 encoded frame, so the pipeline is super simple, something like (properties need to be specified for kvssink):
`gst-launch-1.0 artpecsrc ! video/x-h264,height=720,width=1280,framerate=30/1 ! kvssink`

Let me know if you need more help with this or how far you're able to get and we can go from there.

MushMal, 2020-12-09:
@hassanctech nice! Folks, any further info you need? Please resolve if not

bml1g12, 2020-12-10:
Thank you @hassanctech  and @dfcarpenter - unfortunately it will probably be at least a few weeks until we get the time to do much experimentation on this, so if @MushMal  I am happy to close if that is the expected procedure in this case. 

As I have not used gstreamer or compiled an ACAP before, its hard for me to comment much more until I get the time to do the reading and experimentation into this, but the notes you shared here sound like a great point of reference; I hope if we run into any problems I can reopen this ticket and clarify. 

Thank you!

MushMal, 2020-12-10:
@bml1g12 the reason we try to not keep the issues open/non-transacted is so we can properly prioritize our effort in engaging with them. It's absolutely fine to re-open or cut a new issue and reference the older for more info

bml1g12, 2021-01-20:
> @bml1g12 Yes I can help you with that! There are a couple different approaches you can take, as suggested by @dfcarpenter you can become part of Axis' ACAP program and have access to their embedded [...]

@hassanctech  thank you for your assistance, really appreciated. We setup the AWS Producer SDK on a separate machine to the encoder, and can successfully pull an AXIS rtsp stream into the Producer and send it Kinesis Video Streams. So now it is just a case of getting it working on the encoder itself.

> If gst-inspect-1.0 is on the machine (should already be in path) it will print out the version.

We can ssh onto the encoder, and can access gstreamer interactively. 

> Now you can build our kvssink plugin and as a test you can provide a pipeline via gst-launch-1.0 directly on the command line sending video frames to KVS (not your production solution but it's a good test to verify things are working).

I wanted to clarify on this point, are you suggesting here building a gstreamer on that specific flavour of linux/compilers because it is the same OS as that used on the AXIS encoder; and thus we could build externally using an EC2 instance setup like that (or presumably just a Docker container setup like that) and then use a tool like `scp` to transfer the built gstreamer plugin code onto the encoder ready for use? 

> gst-launch-1.0 directly on the command line sending video frames to KVS (not your production solution but it's a good test to verify things are working).

I also wanted to clarify why you say this would not be production worthy?
If we are using the encoder's native gstreamer install, I presume it would have access to any hardware acceleration from the encoder itself (i.e. their artpecsrc plugin) by default. Is it not production worthy because we have no way of restarting automatically on connection interruptions; something which is managed intelligently by the [callbacks](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-callbacks.html) such as in the AWS provided gstreamer sample code [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/eafb0bf610ce568940e57457217baebb66c23667/samples/kvs_gstreamer_sample.cpp#L243)?

In general, we were wondering the best way to ensure that the stream will infinitely try to reconnect in case of internet failure- as we would like the producer to require little manual intervention once setup on a camera/encoder.

MushMal, 2021-01-20:
Some answers below.

* You will need to cross-compile and then copy the libraries to your target platform
* Production readiness depends on various criteria. I don't see any reasons why not but you will need to work with the hardware platform manufacturer to get your questions answered.
* ContinuousCallbackProvider which is the default provider will attempt to re-connect and re-stream.

I am not sure if @hassanctech will have more info on this but we would try to keep the discussion to KVS specifics as we might not have the latest or most precise answers regarding Axis

bml1g12, 2021-01-21:
> Some answers below.
> 
> * You will need to cross-compile and then copy the libraries to your target platform

I see, I understand now, thank you.

> * Production readiness depends on various criteria. I don't see any reasons why not but you will need to work with the hardware platform manufacturer to get your questions answered.

One thing I am trying to understand is the distinction between sending a Kinesis stream via gstreamer directly:

A)
```
gst-launch-1.0 rtspsrc location=$YOUR_RTSP_STREAM short-header=TRUE user-id=$CCTV_USERNAME user-pw=$CCTV_PASSWORD ! rtph264depay ! h264parse ! kvssink stream-name=$AWS_VIDEO_STREAM_NAME access-key=$AWS_ACCESS_KEY secret-key=$AWS_ACCESS_SECRET_KEY aws-region=$AWS_REGION
```
and via the sample application:
B)
```
AWS_DEFAULT_REGION=$AWS_REGION ./kvs_gstreamer_sample $AWS_VIDEO_STREAM_NAME $YOUR_RTSP_STREAM
```
As both result in a stream to Kinesis. If I understand correctly, one difference is that the sample app includes various mechanisms to improve stream robustness in various cases, such as network failure? I am unclear whether @hassanctech  was also alluding to a benefit here for Method 2 with regards to hardware encoding capability.

> * ContinuousCallbackProvider which is the default provider will attempt to re-connect and re-stream.
> 

I see this sounds like exactly what we would need; does this mean that even if the internet cut out for a very long time (e.g. hours or days), the [sample application](https://github.com/scoville/kinesis/blob/develop/src/kvs_gstreamer_sample.cpp) should restart indefinitely? Sorry if it is a foolish question, but I tried searching on github and AWS Kinesis docs for the strings ContinuousCallbackProvider and ContinuousCallback but could not find any docs or references to this string, so unsure where to look for documentation on how to ensure the stream will reconnect as best as it can indefinately. 

> I am not sure if @hassanctech will have more info on this but we would try to keep the discussion to KVS specifics as we might not have the latest or most precise answers regarding Axis

To give a bit of context on why I think this is relevant to this forum: We wish to scale our product using amazon web services to hundreds of facilities across the company, and currently, the blocker for this is a scalable way to get our video into the cloud. If we cannot install KVS Producer on the encoder; we would need to manage many EC2 instances to run the Producer, which would not be scalable and defeat most of the benefit of using KVS; and indeed we would need to look to non-AWS solutions.

AWS KVS staff have successfully tested KVS Producer SDK on an AXIS P3354 Fixed Dome Network Camera before,  and I understand that information might be precise or up-to-date, but its a great help to understand how this was achieved as it's likely to be very transferable to our case. I have already reached out to AXIS but as the questions, we have at the moment are mainly about understanding how KVS would link in with the hardware of the encoder; and given AXIS do not know anything about KVS, their unlikely to able to support here. 

Very grateful for your support.

MushMal, 2021-01-21:
GStreamer is a framework. gst-launch-1.0 application is just that - an application that allows construction and execution of pipelines.

The difference between a full application and a gst-launch based solution is that in application you are free to do anything you want whereas in gst-launch you can only "fire and forget" pretty much.

Regarding the continuous retry. I would simply suggest running the sample and turning off the network, after a while, turn it back on. It should pick up and stream. There will be dropped frames as the buffer gets overflown.

Installing KVS producer solution on Axis is not something we directly support but I hope that people in the community could help.

hassanctech, 2021-01-21:
@bml1g12 I can definitely assist and getting an application running on the encoder which will stream to KVS.  First question is is your company able to join:  https://www.axis.com/support/developer-support/axis-camera-application-platform.

Once you do this you'll have access to their Axis SDK and cross compile tool chain.  Certain Axis models (yours included) support program installation via this program.  They actually make it very easy for you to build applications for their platform and provide a way for you to programmatically (via the VAPIX web API's that Axis encoders/cameras provide) to install the software on the camera/encoder and to set parameters relative to your application and to start/stop the applications.

You are correct `artpecsrc` will use the hardware encoder, that was not a concern of mine for using GStreamer.

gst-launch provides a way to construct a pipeline on the fly but it's limited, for example you may want a callback function for a particular element, you cannot do this on the command line you would actually need write a C program, there are a plethora of examples out there and I can guide you to some if you need help with this.

The embedded SDK will allow you to access the encoded frame which you can then send to KVS back-end via our Producer C SDK (https://github.com/awslabs/amazon-kinesis-video-streams-producer-c), no GStreamer required, you will have to set up callbacks yourself but you can pretty much copy the callback implementations from what the `kvssink` plugin does.  The real benefit to this is in the case Axis decides one day to remove GStreamer from their cameras/encoders, you will not all of a sudden be out of a solution (it could happen one day when you decided to upgrade your firmware to take in a security patch for example, Axis provides no guarantees that GStreamer will be on there).  However if you wish to take that as an acceptable risk to deal with that when the problem comes (i.e. GStreamer suddenly removed from their cameras), then you would need to compile the `kvssink` plugin and put it on the camera/encoder and load it into your application and construct a simple pipeline.

bml1g12, 2021-01-22:
Thank you for your support. 

> The difference between a full application and a gst-launch based solution is that in application you are free to do anything you want whereas in gst-launch you can only "fire and forget" pretty much.

I see yes, it does seem like calling gstreamer directly would be problematic in that it would not auto-recover from internet downtime; and as I understand also has no native support for sending multiple streams to kinesis (as we have an encoder with 4 cameras attached). So we'll look into Producer SDK as the more robust solution.

> First question is is your company able to join: https://www.axis.com/support/developer-support/axis-camera-application-platform.

Yes we have joined this, although it will take us a few weeks to gain familiarity with the ACAP SDK. 
 
> You can then send to KVS back-end via our Producer C
Would the AWS [C++ flavour ](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/samples)of the producer SDK be equally suitable here? Ah but I see [AXIS recommend C](https://www.axis.com/support/faq/Development/ACAP/FAQ116471) for ACAP, so that's probably why you suggested C here.

As this will now become mostly a C/C++ programming exercise, a colleague of mine (@huysc0ville) who is more familiar with the language will likely be taking us this task over the following weeks.

>  you would actually need write a C program, there are a plethora of examples out there and I can guide you to some if you need help with this.

Yes please, are you referring to [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp) and [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_multistream_sample.cpp) for example? 

As we have 4 cameras per AXIS encoder, it sounds like the multistream example here would be suitable for us, is there an official sample application for the C producer SDK for dealing with multiple streams? We have so far setup a Dockerfile that runs the Producer using the C++ sample, but I notice the samples [in the C repo look a little different](https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/tree/master/samples).   

> The embedded SDK will allow you to access the encoded frame which you can then send to KVS back-end via our Producer C SDK

I personally don't yet have a clear picture in my head how we would pass the AXIS SDK frame into KVS backend Producer SDK, but part of this may be that I am not a confident C developer, and haven't yet studied the AXIS SDK in detail. 

Would it, as an example, be a case of feeding frames from [this AXIS demo ](https://github.com/AxisCommunications/acap3-examples/blob/bcb506ffb2d6c2808ce070365a3ae0549b988ca1/vdostream/app/vdoencodeclient.c#L222) into [this AWS Producer SDK sample](https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/4d39c4f65505ab042d886e64fc6747d35c9217fa/samples/KvsVideoOnlyStreamingSample.c#L142)?

bml1g12, 2021-01-25:
> Are you currently able to extract 4 separate streams from the Axis SDK?

We have yet to start much development on the AXIS side; but I have already inquired to AXIS if they have an example of this we could work from. If they are unclear, I can imagine it would be viable for us to start with an "application" or process running for each camera given the low CPU footprint usage of the current code, although less clear on memory footprint.

Thanks, it sounds like we aught to developing with the AWS Kinesis C SDK but so far have only used C++ one; do you have a sample code for that for use with RTSP stream input? (I appreciate it's not the same code we would need on the AXIS encoder; but before we try and run it on the AXIS encoder, would be good to get the AWS C Kinesis SDK running on a separate machine as we have already done using the C++ SDK.). I think [this sample](https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/samples/KvsVideoOnlyStreamingSample.c) is based on reading from frames on disk if I understand correctly.

disa6302, 2021-02-16:
@bml1g12 , 

You are correct, the sample is based on reading from frames on disk. Unfortunately, we do not have a C producer sample with RTSP camera.

bml1g12, 2021-02-17:
OK that's a shame, thank you for this information. I will close this issue for now.

closed 2021-02-17 by bml1g12