awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #624: [QUESTION] Any Insights with DNS resolve timeouts

suggestedfixes, 2020-12-04:
```
2020-12-04 19:43:32 ERROR   blockingCurlCall(): Curl perform failed for url https://${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com/role-aliases/1C8259188943/credentials with result Timeout was reached : Resolving timed out after 3511 milliseconds
```
But when we do a 
nslookp ${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com
```
# nslookup -debug ch4n7f72w5ltt.credentials.iot.us-east-2.amazonaws.com
Server:         8.8.8.8
Address:        8.8.8.8:53

Query #0 completed in 209ms:
Non-authoritative answer:
Name:   ${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com
Address: 3.128.77.168
Name:   ${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com
Address: 3.19.113.0
Name:   ${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com
Address: 18.218.205.169

Query #1 completed in 2532ms:
Non-authoritative answer:
Name:   ${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com
Address: 2600:1f00:6000::12da:cda9
Name:   ${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com
Address: 2600:1f00:6000::380:4da8
Name:   ${CREDENTIALS_ENDPOINT}.credentials.iot.us-east-2.amazonaws.com
Address: 2600:1f00:6000::313:7100

```

It's getting back an ipv4 answer in 426ms and ipv6 answer in 2532 ms, it might be that the ipv6 is approaching the timeout. I wonder if there is anything we can do to disable ipv6 from the SDK, or increase IoT credentials timeouts?

Apologize ahead of time as I'm not familiar with the codebase.

suggestedfixes, 2020-12-04:
Just to add a stir to the mix, the IoT credentials provider works on the same device for WebRTC

suggestedfixes, 2020-12-04:
Did some tracing:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/7f4ea67b667217670cd90a6bc119e11dd6f12f9b/src/source/Common/IotCredentialProvider.h#L11

disa6302, 2021-01-18:
Closing since this ticket has gone stale.

closed 2021-01-18 by disa6302