awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #625: Kvssink and webrtc [QUESTION]

Cocoont, 2020-12-07:
Hi, 

I use the gstreamer plugin kvssink to stream my raspberry camera but there is like a 7s latency when I see the camera on https://aws-samples.github.io/amazon-kinesis-video-streams-media-viewer/. I want to suppress that. 
I saw that there is Low-latency two-way media streaming with WebRTC. Is it possible to use webrtc with kvssink ?

Thank you for your help

MushMal, 2020-12-07:
HLS/DASH/Console playback are all "live" playbacks and are not "realtime". Live playback is based on the indexed/persisted fragments which means that you will have at least a fragment duration size latency. You can build your own realtime player based on GetMedia API that is a realtime API. WebRTC provides another P2P realtime streaming capability. As of now, we don't have a GStreamer plugin for the WebRTC.

You could build a GStreamer application to handle your media source and then feed the frames into the WebRTC SDK similar to the GStreamer sample application that we have in the repository.

Cocoont, 2020-12-07:
Ok thanks for your response !

closed 2020-12-07 by Cocoont