awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #628: [FEATURE] Add specific getting started documentation for raspberry pi

disa6302, 2020-12-07:
**Is your feature request related to a problem? Please describe.**
The current raspberry pi documentation seems broken. It is important to update the documentation so that users do not spend a lot of time trying to get our SDKs running.

**Describe the solution you'd like**
It is easier to get started with our SDK if there is a specific getting started doc for raspberry pi instead of having "if raspberry pi, do this...".

faustinopalma, 2020-12-09:
So far, after spending over 4 hours, I am still not able to get this running on a Raspberry pi 4. I tried multiple times even from a new Raspberry image with no success. Is there any documentation which gives a step by step guide that works?

MushMal, 2020-12-09:
@faustinopalma this is very specific feature request. Can you cut a new issue with detailed description of what's "not working" and provide more context. RPi 4 is a supported platform. The SDK builds just fine and the samples run just fine. You might want to check out the readme.md at the root of the repository and also look at the RPi specifics and troubleshooting here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md

disa6302, 2020-12-09:
@faustinopalma ,

I will be adding this soon. That is what this ticket is for. But can you be specific on what does not work?

faustinopalma, 2020-12-09:
I first tried following this documentation:
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html
which does not work at the point when it asks to run the "install-script" file because the file is not existing.

Then I tried following the steps at the base repository:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp
but I get the error: No such element or plugin 'kvssink'

I verified that the GST_PLUGIN_PATH variable is set correctly:
pi@raspberrypi:~ $ echo $GST_PLUGIN_PATH
/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/build

and the compilation actually completed without errors.

disa6302, 2020-12-09:
Can you verify `libgstkvssink.so` exists in the build directory? 
Did you run: `cmake .. -DBUILD_GSTREAMER_PLUGIN=ON`?

faustinopalma, 2020-12-09:
> @faustinopalma ,
> 
> I will be adding this soon. That is what this ticket is for. But can you be specific on what does not work?
> 
> Also, there is one missing instruction in README.md:
> sudo apt-get install gstreamer1.0-omx
> 
> Can you run this as well and see if it solves the problem for you?

I tried but GStreamer was already installed and nothing changed. I still get the error: No such element or plugin 'kvssink', despite the compilation ended without errors.

disa6302, 2020-12-09:
Can you verify `libgstkvssink.so` exists in the build directory?
Did you run: `cmake .. -DBUILD_GSTREAMER_PLUGIN=ON`?

Can you post contents of your `build` directory?

faustinopalma, 2020-12-09:
> Can you verify `libgstkvssink.so` exists in the build directory?
> Did you run: `cmake .. -DBUILD_GSTREAMER_PLUGIN=ON`?

No, I didn't use the flag '-DBUILD_GSTREAMER_PLUGIN=ON'
I just run the command: 'cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE' following the doc and now it works.
Thanks!

closed 2020-12-10 by disa6302